lgsvl/simulator #912: ROS node controls throttle, brake, gear, and steering?

manonechu, 2020-08-13:
Hi, thank you for the great simulator.

I would like to develop a ROS node to control or publish the throttle, brake, gear and steering commands to the simulator so that the vehicle in the simulator can move based on these commands. Could you advise me what are the topic names and message types of these commands used by the simulator? 

By the way, I use C++ to code. Do you have a relevant example or tutorial for me to follow about the above question?

Many thanks!

hadiTab, 2020-08-13:
@manonechu I would recommend using the [lgsvl_msgs](https://github.com/lgsvl/lgsvl_msgs) package. The most convenient way to install it is through apt:
```
sudo apt update
sudo apt install ros-$ROS_DISTRO-lgsvl-msgs
```

Specifically, the [`VehicleControlData`](https://github.com/lgsvl/lgsvl_msgs/blob/master/msg/VehicleControlData.msg) message type is what you seem to be looking for.

As for tutorials, [this guide](http://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber%28c%2B%2B%29) should cover everything you need for writing a publisher node in ROS.

manonechu, 2020-08-13:
@hadiTab Thank you for your reply. I will be coding a ROS cpp node to control the vehicle. I will use the VehicleControlData message type for sure. My further question is, I need to publish this message on a topic name, can I name the topic by myself or are there any requirements to name the topic?
I appreciate your kind help.

hadiTab, 2020-08-13:
@manonechu you can name the topic yourself and specify it in the vehicle sensor configuration in the simulator web UI. [Here](https://www.lgsvlsimulator.com/docs/sensor-json-options/#vehicle-control) are example parameters for the vehicle control sensor.

manonechu, 2020-08-13:
@hadiTab Got it. Thank you very much. I think the steps to control the vehicle using a ROS node are: 1. run the simulator. 2. launch the ros_bridge. 3. rosrun the ROS node (I use c++). If these are not right, please kindly advise me.
By the way, instead of using normal sedans or SUVs, is there semi-truck available, too? I would like to simulate a semi-truck.
Thank you again.

hadiTab, 2020-08-13:
@manonechu that sounds about right. As for semi-truck, unfortunately we don't have an ego vehicle semi-truck available right now.

manonechu, 2020-08-13:
@hadiTab Thanks a lot for all your answers!

manonechu, 2020-08-14:
@hadiTab I would like to ask one more question on the waypoints for a vehicle to follow: how should I generate or select these waypoints on the map and then pass them to the vehicle to follow? Is there a tool available to do that?
By the way, I would use roscpp to develop nodes.
Thank you again!

hadiTab, 2020-08-15:
The simulator has waypoints annotated onto the lanes which can be exported as HD maps in several formats. I think the easiest way would be to use one of the supported HD maps. @luqiang21 might be able to help more with this.

luqiang21, 2020-08-15:
@manonechu As @hadiTab said, NPC can follow map annotations. And you can either annotated them yourself using simulator's `HD Map Annotation` window or import from any of supported map formats: `lanelet2`, `apollo opendrive`, or `standard opendrive`.

manonechu, 2020-08-20:
Hi @hadiTab, is there any way to create and import a semi-truck model by myself to the simulator? Your advice would be appreciated.

hadiTab, 2020-08-20:
@EricBoiseLGSVL

EricBoiseLGSVL, 2020-08-20:
@manonechu Yes, you can add a truck to the list of NPCs in the simulator but you will have issues with the trailer behavior.  You will need a Fixed Joint component to attach the trailer to the NPC.  We are working on this now but it won't be released soon.

EricBoiseLGSVL, 2020-08-21:
@manonechu Be sure to add the new prefab to the list in Assets/Resources/NPCSettings.asset and look at how other NPC meshes/materials are created.  If you copy the names of the meshes and materials you will be able to add a new vehicle.  You may need to alter some code to handle multi-wheeled rig

manonechu, 2020-08-21:
Hi @EricBoiseLGSVL, thank you very much for your instructions. I am trying to install the latest version of the simulator, but I am having an issue to run the "simulator" executive file. By following the instruction [here](https://www.lgsvlsimulator.com/docs/getting-started/) for Ubuntu, right after I run the "simulator" executive file, a black screen pops out but then it disappears, and the simulator does not come out. My issue seems the same as [here](https://github.com/lgsvl/simulator/issues/527). I have worked on solving this issue for a few hours but I failed to run the simulator. 
By the way, I am using my laptop that has GTX 1050 Ti GPU. Is it because my GPU is too old? Please kindly advise me. Thank you very much again.

EricBoiseLGSVL, 2020-08-21:
That is an older GPU, We recommend an 8gb GPU at least.  Can you post your player.log so we can see the error please? Also nvidia-smi info

EricBoiseLGSVL, 2020-08-24:
Thanks @lemketron 
@manonechu Does this solve your issue?

manonechu, 2020-08-25:
Thank you very much @lemketron and @EricBoiseLGSVL. It works now!
I really appreciate your help.

EricBoiseLGSVL, 2020-08-25:
@manonechu Glad to help.  Thanks for posting to help others that might have the same issue.

closed 2020-08-25 by EricBoiseLGSVL

rahul14k14, 2020-09-06:
> @hadiTab Got it. Thank you very much. I think the steps to control the vehicle using a ROS node are: 1. run the simulator. 2. launch the ros_bridge. 3. rosrun the ROS node (I use c++). If these are [...]

Hi @manonechu,
I am also trying to do the same but I am newbie in this so can you please help regarding the same by providing the sample ros node of cpp or Python so that I can try the same. Also if you can provide me any document for better understanding. I am very thankful to you.

EricBoiseLGSVL, 2020-09-08:
@rahul14k14 Did you follow @hadiTab links above?  What are you exactly having issues with?  Simulator or ROS?

rahul14k14, 2020-09-10:
Hi @EricBoiseLGSVL,
Thank you for the reply.
I had followed the above links.
In LG simulator I am able to see SUB node for vehicle_control_cmd so I was trying to publish the data for that node but the count is still 0 for that.
Can you please help me.
How I can control vehicle in lg simulator over ros2. Is there any code available.
Waiting for your reply.
Thanks in advance.

EricBoiseLGSVL, 2020-09-10:
It sounds like you are having issues running the ROS node.  Please read the ROS documentation on starting a ROS node [here](http://wiki.ros.org/ROS/Tutorials/UnderstandingNodes).  Since this is a closed issue, please create a new one if this information does not solve the problem you are having.  This makes it easier for other users to search issues.