lgsvl/simulator #913: Fisheye Camera flashing？

jiayiyi1991, 2020-08-14:
When I run project，fisheye flashing like this picture :
[
[image]
](url)

Fisheye Params :
 {
    "type": "Color Camera",
    "name": "Fish Camera",
    "params": {
      "Width": 1920,
      "Height": 1080,
      "Frequency": 15,
      "JpegQuality": 75,
      "FieldOfView": 50,
      "MinDistance": 0.1,
      "MaxDistance": 1000,
      "Topic": "/simulator/fish_camera",
      "Frame": "camera",
      "Distorted": true,
      "DistortionParameters": [
        -0.25349, 0.11868, 0, 0
      ],
  "Fisheye": true,
      "Xi":1,
"CubemapSize":1024
    },
    "transform": {
      "x": 0,
      "y": 1.7,
      "z": -0.2,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
}

jiayiyi1991, 2020-08-14:
Fisheye Camera Inspector:
[image]

rongguodong, 2020-08-14:
It is a known issue. We are investigating a solution for this.

jiayiyi1991, 2020-08-14:
> It is a known issue. We are investigating a solution for this.

ok,thank you very much!
Do you use other asset for fisheye now?
like assetstore:
https://assetstore.unity.com/packages/vfx/shaders/fullscreen-camera-effects/dome-tools-62664

rongguodong, 2020-08-14:
No, we developed our own distortion, which matches the calibration model used by Apollo.

Most of fisheye effects online are not following any calibration models. But I will take a look of the one your provided to learn more about it. Thanks for the information!

jiayiyi1991, 2020-08-17:
@rongguodong  thank you, Looking forward to fisheye update. 😁

jiayiyi1991, 2020-08-17:
@rongguodong 
I find LensDistortion function can run in built-in render pipeline.
Main Camera:
[image]
Dome Tools:
[image]
Lgsvl LensDistortion(xi:1, DistortionParameters:  -0.25349f, 0.11868f, 0, 0):
[image]
Is this right?

rongguodong, 2020-08-17:
What do you mean by "LensDistortion function can run in built-in render pipeline"?

jiayiyi1991, 2020-08-18:
> What do you mean by "LensDistortion function can run in built-in render pipeline"?
I mean the default render pipeline before "HDRP & SRP".
I create a built-in render pipeline project, than create a fisheye camera by "lgsvl CameraSensorBase" code:

` void Start()
    {
        SensorCamera = GetComponent<Camera>();
        len = new LensDistortion(LensDistortionShader);
        var DistortionParameters = new List<float> { -0.25349f, 0.11868f, 0, 0 };
        var Xi = 1;
        var FrustumHeight = 2 * Mathf.Tan(FieldOfView / 2 * Mathf.Deg2Rad);
        var FrustumWidth = FrustumHeight * Width / Height;
        len.InitDistortion(DistortionParameters, FrustumWidth, FrustumHeight, Xi);
        CubemapTexture = new RenderTexture(CubemapSize, CubemapSize, 24, RenderTextureFormat.ARGB32, RenderTextureReadWrite.sRGB)
        {
            dimension = UnityEngine.Rendering.TextureDimension.Cube,
            antiAliasing = 1,
            useMipMap = false,
            useDynamicScale = false,
            wrapMode = TextureWrapMode.Clamp,
            filterMode = FilterMode.Bilinear,
        };
        CubemapTexture.Create();
        DistortedTexture = new RenderTexture(Width, Height, 24, RenderTextureFormat.ARGB32, RenderTextureReadWrite.sRGB)
        {
            dimension = UnityEngine.Rendering.TextureDimension.Tex2D,
            antiAliasing = 1,
            useMipMap = false,
            useDynamicScale = false,
            wrapMode = TextureWrapMode.Clamp,
            filterMode = FilterMode.Bilinear,
            enableRandomWrite = true,
        };
        DistortedTexture.Create();
        Target.texture = DistortedTexture;
    }

    private void LateUpdate()
    {
        SensorCamera.stereoSeparation = 0f;
        SensorCamera.RenderToCubemap(CubemapTexture, 63, Camera.MonoOrStereoscopicEye.Left);
        len.UnifiedProjectionDistort(CubemapTexture, DistortedTexture);
    }`

jiayiyi1991, 2020-08-18:
> What do you mean by "LensDistortion function can run in built-in render pipeline"?

I mean the default render pipeline before "HDRP & SRP".
I create a built-in render pipeline project, than create a fisheye camera by "lgsvl CameraSensorBase" code:

`    void Start()
    {
        SensorCamera = GetComponent<Camera>();
        len = new LensDistortion(LensDistortionShader);
        var DistortionParameters = new List<float> { -0.25349f, 0.11868f, 0, 0 };
        var Xi = 1;
        var FrustumHeight = 2 * Mathf.Tan(FieldOfView / 2 * Mathf.Deg2Rad);
        var FrustumWidth = FrustumHeight * Width / Height;
        len.InitDistortion(DistortionParameters, FrustumWidth, FrustumHeight, Xi);
        CubemapTexture = new RenderTexture(CubemapSize, CubemapSize, 24, RenderTextureFormat.ARGB32, RenderTextureReadWrite.sRGB)
        {
            dimension = UnityEngine.Rendering.TextureDimension.Cube,
            antiAliasing = 1,
            useMipMap = false,
            useDynamicScale = false,
            wrapMode = TextureWrapMode.Clamp,
            filterMode = FilterMode.Bilinear,
        };
        CubemapTexture.Create();
        DistortedTexture = new RenderTexture(Width, Height, 24, RenderTextureFormat.ARGB32, RenderTextureReadWrite.sRGB)
        {
            dimension = UnityEngine.Rendering.TextureDimension.Tex2D,
            antiAliasing = 1,
            useMipMap = false,
            useDynamicScale = false,
            wrapMode = TextureWrapMode.Clamp,
            filterMode = FilterMode.Bilinear,
            enableRandomWrite = true,
        };
        DistortedTexture.Create();
        Target.texture = DistortedTexture;
    }`
`    private void LateUpdate()
    {
        SensorCamera.stereoSeparation = 0f;
        SensorCamera.RenderToCubemap(CubemapTexture, 63, Camera.MonoOrStereoscopicEye.Left);
        len.UnifiedProjectionDistort(CubemapTexture, DistortedTexture);
    }`

EricBoiseLGSVL, 2020-08-18:
@jiayiyi1991 Yes, the older Unity render pipeline had this functionality but things are different with HDRP.  We are looking into the exposure issues with a fisheye camera effect.

jiayiyi1991, 2020-08-19:
@EricBoiseLGSVL  thank you! I‘ll do some tests by old pipleline, and study on Lens Distortion simulation.

EricBoiseLGSVL, 2020-08-19:
Sounds good.  We are looking into it also.  We have found that Unity is getting different exposure values at different locations, you will need to force HDRP exposure to be a constant

jiayiyi1991, 2020-08-19:
> Sounds good. We are looking into it also. We have found that Unity is getting different exposure values at different locations, you will need to force HDRP exposure to be a constant

yes, It looks better: 
(fov270, xi:1, DistortionParameters: -0.25349f, 0.11868f, 0, 0)
[image]

closed 2020-08-25 by jiayiyi1991

sanjay1794, 2020-09-19:
hi can any one help me how to setup dome tools ??

EricBoiseLGSVL, 2020-09-21:
@sanjay1794 Can you explain dome tools please?  @jiayiyi1991 set the sensor config parameters and made an adjustment to the HDRP post processing effect Exposure.