lgsvl/simulator #915: Mac OS Support

sachit-tri, 2020-08-14:
Hi,

First of all, thanks for your great work on developing this simulator.

I was wondering if there was any planned support for MacOS in the future? If not, what would it involve to support it? I noticed I was able to build this on my Mac, but looks like there are some shader issues when running a simulation. 

I ask because a lot of individuals in my organization are MacOS users. It'd be really nice if we could build and run this simulator on our Macs instead of dual booting / Remote Desktop to some other machines.

EricBoiseLGSVL, 2020-08-14:
@sachit-tri  Glad you are finding LGSVL Simulator useful.  We don't have Mac support on the roadmap right now.  We'll have to see how much work this will take.

rongguodong, 2020-08-14:
I see Unity now has Mac version. You may try to build our simulator on Mac to see if it works.

Please share you findings here. :)

sachit-tri, 2020-08-14:
Thanks for the responses @EricBoiseLGSVL and @rongguodong . I did try building it, but it seems like at runtime (San Francisco using the Jaguar), this is what I see when I run a simulation:

[image]

It's pink all over so I'm assuming there's something not working correctly with the shaders.

martins-mozeiko, 2020-08-16:
Unity had Mac version since forever - it was originally released as Mac exclusive.

For SF and CubeTown or other environments and vehicles you will need to adjust [code that builds assets](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Editor/Build.cs#L334). As currently code builds assets only for Windows and Linux. Then adjust the [loading code](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Web/Loader.cs#L576) to load correct version. Shaders in asset bundles are platfrom specific - that's why each asset bundle contains multiple copies of them. I suggest searching rest of code for "Windows" and "Linux" to check what needs to be adjusted.

To fully run all simulator features, you will need to also build few native plugins - as currently there are shared libraries only for Windows and Linux. This includes: [JpegEncoder](https://github.com/lgsvl/simulator/tree/master/Assets/Plugins/JpegEncoder), [PngEncoder](https://github.com/lgsvl/simulator/tree/master/Assets/Plugins/PngEncoder), [SQLite](https://github.com/lgsvl/simulator/tree/master/Assets/Plugins/SQLite) and [LASzip](https://github.com/lgsvl/simulator/tree/master/Assets/Plugins/LASzip). These folders have shell scripts to build for Linux, they probably will work for Mac just fine, but you'll need to change extension from .so to .dylib.

sachit-tri, 2020-08-17:
Thanks for that detailed response @martins-mozeiko. 

> But because there are not so many users with Mac, and also performance on Mac is not good (as there are no good GPU's on most of Mac machines) we kind of dropped supporting it.

Thanks for clarifying that too, it makes a lot of sense.

closed 2020-10-20 by EricBoiseLGSVL

schissmantics, 2021-08-11:
> Unity had Mac version since forever - it was originally released as Mac exclusive.
> 
> For SF and CubeTown or other environments and vehicles you will need to adjust [code that builds assets](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Editor/Build.cs#L334). As currently code builds assets only for Windows and Linux. Then adjust the [loading code](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Web/Loader.cs#L576) to load correct version. Shaders in asset bundles are platfrom specific - that's why each asset bundle contains multiple copies of them. I suggest searching rest of code for "Windows" and "Linux" to check what else needs to be adjusted.
> 
> To fully run all simulator features, you will need to also build few native plugins - as currently there are shared libraries only for Windows and Linux. This includes: [JpegEncoder](https://github.com/lgsvl/simulator/tree/master/Assets/Plugins/JpegEncoder), [PngEncoder](https://github.com/lgsvl/simulator/tree/master/Assets/Plugins/PngEncoder), [SQLite](https://github.com/lgsvl/simulator/tree/master/Assets/Plugins/SQLite) and [LASzip](https://github.com/lgsvl/simulator/tree/master/Assets/Plugins/LASzip). These folders have shell scripts to build for Linux, they probably will work for Mac just fine, but you'll need to change extension from .so to .dylib.
> 
> After all this simulator will run fine. We had it running on Mac at some point just fine. But because there are not so many users with Mac, and also performance on Mac is not good (as there are no good GPU's on most of Mac machines) we kind of dropped supporting it.

How to change `.so` extensions to `.dylib`? I downloaded unity hub which I used to load the project, and then I choose simulator > build from the menu, which produces an executable. What should I do to ensure I won't get these funny colors (for the project to work properly)?

EricBoiseLGSVL, 2021-08-11:
You will need to change our build code to also support Mac. We highly recommend you do not use a Mac for the simulator. You are going to run into many issues with Unity shaders and our sensor/environment shader graphs.

EricBoiseLGSVL, 2021-08-11:
@schissmantics good luck and thanks for using the simulator. Be sure to post any findings here if you discover any solutions.

schissmantics, 2021-08-11:
I'm just not sure about how the code should be modified, so if there is some guide I should follow, please let me know.

EricBoiseLGSVL, 2021-08-11:
Not on this, it will be trial and error.  Mostly look at Build.cs.  You can see logic for Windows or Linux and add Mac to it.  I am worried that lidar/depth/seg/road shader sensors will fail because they were never written for Macs.  Once you make sure bundles are covered, then see the next errors or issues.

Zhangmaopeng88, 2022-05-03:
thanks，actually I also running the code in the unity on my Mac Pro ,but's it's a pity that lgsvl could not running in the Mac right now.