lgsvl/simulator #917: About elevationProfile and superelevation

yumianhuli2, 2020-08-16:
Hello LgsvlTeam!
I have two questions in this post.
1 CalculateLinePoints and CalculateArcPoints have coordinates y that compute the elevation, but why does CalculateSpiralPoints not compute the coordinate y of this elevation?

2 Superelevation - I have not found the code to calculate the tilt Angle of the cross section. Do you know the formula and method for calculating it?
Thanks！

luqiang21, 2020-08-18:
@yumianhuli2 
1. This is a known bug, `CalculateSpiralPoints(..)` and `CalculatePoly3Points(..)` also need to compute elevation. We have fixed it internally, you will see it in next release.
2. You can refer opendrive definition document for the formula of calculating it.
3. All points are computed assuming `0` angle. Finally we need to rotate them by the given heading.

yumianhuli2, 2020-08-19:
CalculateLinePoints()--This is probably not the right function, it doesn't use angles to compute the coordinates，Especially the Z is always Zero.

luqiang21, 2020-08-19:
@yumianhuli2 As you can in this [line](https://github.com/lgsvl/simulator/blob/8b02c852c0642347f0b05125c726ceb09bfa0b26/Assets/Scripts/Editor/OpenDriveMapImporter.cs#L277), it is using `hdg` angle to compute the coordinates. `z` is 0 because it first computes without rotation along `x` direction, then rotates the coordinate to get correct result.

yumianhuli2, 2020-08-19:
> @yumianhuli2 As you can in this [line](https://github.com/lgsvl/simulator/blob/8b02c852c0642347f0b05125c726ceb09bfa0b26/Assets/Scripts/Editor/OpenDriveMapImporter.cs#L277), it is using `hdg` angle [...]

Hi! luqiang!
pos = Quaternion.Euler(0f, -(float)(geometry.hdg * 180f / Math.PI), 0f) * pos 
You can only change the rotation but you can't change the coordinates？？？

Thanks！

luqiang21, 2020-08-19:
@yumianhuli2 

1. What do you mean by `can only change the rotation but can't change the coordinates`? There is no rotation for each `pos`, it is only a 3D coordinate with `x`, `y`, `z`.

2. In `CalculateParamPoly3Points()`, following two lines are converting from `UV` space to `XY` space.
```
                pos = Quaternion.Euler(0f, -(float)(geometry.hdg * 180f / Math.PI), 0f) * pos;
                points.Add(origin + pos);
```

yumianhuli2, 2020-08-20:
> @yumianhuli2
>
>     1. What do you mean by `can only change the rotation but can't change the coordinates`? There is no rotation for each `pos`, it is only a 3D coordinate with `x`, `y`, `z`.
>
>     2. [...]

OK!Hope for updating elevation bug.Thanks

closed 2020-08-20 by yumianhuli2

luqiang21, 2020-08-24:
@yumianhuli2 You already asked this https://github.com/lgsvl/simulator/issues/931 right? I am checking it.