lgsvl/simulator #918: [cyber] protoc version for protobuf c# outputs

ntutangyun, 2020-08-17:
Hi team,

I'd like to modify some of the `proto` definition from Apollo's side. And I need to update the `protobuf` C# compiled output for LGSVL as well. 

May I know what version of `protoc` and command you are using to compile the proto into the folder: 
`/Assets/Scripts/Bridge/Cyber/Protobuf`

thanks.

ntutangyun, 2020-08-17:
@martins-mozeiko @daviduhm Please help. thanks.

ntutangyun, 2020-08-17:
seems that you're using `protogen` by protobuf-net.Protogen.  May I know which version are you using ? thanks

ntutangyun, 2020-08-17:
I tried to use `protogen` version `1.0.10`, but it shows `Cannot implicitly convert type double to float` as shown below:

[image]

ntutangyun, 2020-08-17:
found similar [issue ](https://github.com/lgsvl/simulator/issues/207)

ntutangyun, 2020-08-17:
I tried and failed to build the `protogen`... 
I've install all the required `.netframework` but the command `dotnet build` failed to find them

Could you provide the built version in the `lgsvl/protobuf-net` repo ? thanks.

rongguodong, 2020-08-17:
https://protogen.marcgravell.com/protogen/protogen%202.3.16.zip
Please try this download link.

ntutangyun, 2020-08-17:
@rongguodong thanks. seems to be working with the following command:

```bash
protogen.exe --proto_path=. --csharp_out=protobuf_out/ +names=original +langver=3 modules/perception/proto/perception_obstacle.proto
```

ntutangyun, 2020-08-17:
@rongguodong Also, I noticed that the proto files have been updated in Apollo's `pre6` branch. Are there plans to compile the latest proto files?

rongguodong, 2020-08-17:
Yes, our next release should be compatible with pre6

closed 2020-08-18 by ntutangyun