lgsvl/simulator #919: Run the simulation in headless mode

joshwapohlmann, 2020-08-17:
## Feature Request
Enable an option to **Run the simulation in headless mode** without an X11 session nor an attached display.

## Motivation
We want to run system level integration tests with our software stack witout being dependant of an attached display.
But currently, there is no way to run LGSVL without a valid $DISPLAY or in other way associated X11 session as far as I know.

## Problem
1. I tried turning on the built option [EnabledHeadlessMode](https://docs.unity3d.com/ScriptReference/BuildOptions.EnableHeadlessMode.html).
2. This (surprisingly) works** until I need any rendered output from a camera sensor, which will cause errors, e.g. from the CameraSensor (without distortions)

```Failed to read GPU texture 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 35)

This GfxDevice does not support asynchronous readback 
(Filename:  Line: 1725)
```
----
** The test passes which spawns the ego vehicle, issues a drive command and tests if a `HallSensor` indicates movement.

EricBoiseLGSVL, 2020-08-17:
@joshwapohlmann Yes, if you require camera images you must render to a display.  We are looking into solutions for this.

martins-mozeiko, 2020-08-17:
Yeah, Unity doesn't allow to use GPU without X11. Any GPU related functionality won't work if you build headless mode - that includes camera and lidar sensors. We cannot do much here, that's for Unity to fix. So you need to use X11. But there's no need to attach display (as in monitor) - you just need running valid X11 session with GPU driver installed.

joshwapohlmann, 2020-08-18:
@martins-mozeiko This is what we concluded to for the time being, which is also partly confirmed [here](https://answers.unity.com/questions/1672109/native-texture-pointer-returns-0-in-headless-build.html).

This makes building an automatic pipeline very hard, but there should be a way. 
> you just need running valid X11 session with GPU driver installed.

thanks for the hint, I look into this.

@EricBoiseLGSVL should this issue be closed due to "cannot be done with unity atm" or stay open due to "may be possible in the future"?

EricBoiseLGSVL, 2020-08-18:
@joshwapohlmann Let's close it because we don't have any idea if or when Unity will support.

closed 2020-08-18 by EricBoiseLGSVL