lgsvl/simulator #920: How to remove network account register instead of using local data

iamshane, 2020-08-18:
Hello, I have been research on this simulator for a while, and I realize that the the simulator must log in on the specified website to register. 
I am wondering that I can remove the website log in organism,  instead of loading all data in local? Thank you.

EricBoiseLGSVL, 2020-08-18:
@iamshane  What version of the simulator are you using and what are you trying to do? Local data?  Can you explain a bit more so we can help?

iamshane, 2020-08-18:
@EricBoiseLGSVL @zelenkovsky 
 I am using the version of 202005 which is the latest version.
Now the LGSVL simulator must log in via website. How can I make it working that do not need to register and log in?  Let me explain it with details, if run the simulator, then it will jump on another website port:8080 to register and log in, I just want to change the log in organism without register and log in, and it can be configured the data via UI, change the car apperance, the map and so on. How can I make it? thank you

zelenkovsky, 2020-08-26:
Requirement to register and maintain logged in status is intentional, we expect to provide more cloud based features to the simulator in very near future. It would include running cloud simulation at scale, content sharing and etc.

While running continuously improving open source version of the simulator locally will be always free, most of value added premium features will be delivered via the cloud.

Our current license does not prevent modifying the source code for debugging and development of your AD system (which we consider a non commercial activity).

PS: Please close this issue if this answers your question.

closed 2020-10-20 by EricBoiseLGSVL