lgsvl/simulator #921: Lgsvl imports PCD in unity and displays all black

wisdom-bob, 2020-08-18:
the error as show below:
Point Cloud has no color and intensity data. Everything will be black!
UnityEngine.Debug:LogError(Object)
Simulator.Editor.PointCloud.Trees.PointImportJobs:GetColorAccess(List`1, Byte*, Int32) (at Assets/Scripts/Editor/PointCloud/Trees/PointImportJobs.cs:781)
Simulator.Editor.PointCloud.Trees.PointImportJobs:ConvertData(MemoryMappedViewAccessor, PointCloudPoint[], List`1, TransformationData, Int32, Int64, String) (at Assets/Scripts/Editor/PointCloud/Trees/PointImportJobs.cs:597)
Simulator.Editor.PointCloud.Trees.PointProcessor:ConvertPointsDefault(DefaultHeaderData, NodeProcessorDispatcher, TransformationData) (at Assets/Scripts/Editor/PointCloud/Trees/PointProcessor.cs:163)
Simulator.Editor.PointCloud.Trees.PcdPointProcessor:ConvertPoints(NodeProcessorDispatcher, TransformationData) (at Assets/Scripts/Editor/PointCloud/Trees/PcdPointProcessor.cs:192)
Simulator.Editor.PointCloud.Trees.NodeTreeBuilder:BuildNodeTree(TreeImportSettings) (at Assets/Scripts/Editor/PointCloud/Trees/NodeTreeBuilder.cs:77)
Simulator.Editor.PointCloudImporterWindow:DrawImportButton() (at Assets/Scripts/Editor/PointCloudImporterWindow.cs:502)
Simulator.Editor.PointCloudImporterWindow:OnGUI() (at Assets/Scripts/Editor/PointCloudImporterWindow.cs:211)
UnityEngine.GUIUtility:ProcessEvent(Int32, IntPtr)

The .pcd file can be opened correctly in maptoolbox(unity). It's not a bad file. However, when import this file into LGSVL, something goes wrong.
How to cover this problem, I need someone help me.. Thanks a lot.

kamkolak, 2020-08-19:
Hey @wisdom-bob 
This seems to be an issue with our PCD importer, that doesn't properly register RGB field in PCD file. Before the fix is released, I can suggest converting your point cloud to PLY format (PLY importer doesn't have this issue), or you could temporarily modify your local PcdPointProcessor.cs file to handle the RGB field.

I can't be sure abot your file exact format without having access to it, but adding the snippet below to on line 167 in PcdPointProcessor.cs might resolve the issue:
```
if (fields[i] == "rgb" && elementSize == 4)
{
    result.Elements.Add(new PointElement() {Type = PointElementType.Byte, Name = PointElementName.R, Offset = elementOffset});
    result.Elements.Add(new PointElement() {Type = PointElementType.Byte, Name = PointElementName.G, Offset = elementOffset + 1});
    result.Elements.Add(new PointElement() {Type = PointElementType.Byte, Name = PointElementName.B, Offset = elementOffset + 2});
}
```

wisdom-bob, 2020-08-20:
> Hey @wisdom-bob
> This seems to be an issue with our PCD importer, that doesn't properly register RGB field in PCD file. Before the fix is released, I can suggest converting your point cloud to PLY [...]

> Hey @wisdom-bob
> This seems to be an issue with our PCD importer, that doesn't properly register RGB field in PCD file. Before the fix is released, I can suggest converting your point cloud to PLY [...]

I choose your second idea. Now, the pcd can be correctly imported. But I can't bring the rgb field into the it. thanks for your idea. I will try your code.

closed 2020-09-16 by EricBoiseLGSVL