lgsvl/simulator #923: Lgsvl Simulator Npc spawn

asifsid-32, 2020-08-19:
Hello i would like to spawn multiple npc at different locations in the map autonomous stuff. To be specific the NPC need to be spawned at different parking lots.

How is it possible to do that ? With the available python api's they are getting spawned at random places. I need to get exact coordinates of parking lots.

Is it possible to do that ?

ntutangyun, 2020-08-19:
A relatively simple solution is that you could drive the EGO vehicle to the exact location where you want NPC vehicle to spawn, record the EGO vehicle's transform (position + rotation) information. And use this information to spawn the NPC vehicles.

EricBoiseLGSVL, 2020-08-19:
@asifsid-32 Be sure to not use Random Traffic option and spawn the NPC's with the API.  @ntutangyun is correct, use the EGO to get the position.  We also have SF map that has parking lots.  This map is open source so you can open in the editor and just place NPCs models (remove the scripts and add collider).  Then make a new bundle and you have the cars parked correctly from the start, no need to use API

asifsid-32, 2020-08-20:
> @asifsid-32 Be sure to not use Random Traffic option and spawn the NPC's with the API. @ntutangyun is correct, use the EGO to get the position. We also have SF map that has parking lots. This map is [...]

Can i get any youtube tutorial for the editing this SF map ?

asifsid-32, 2020-08-20:
> A relatively simple solution is that you could drive the EGO vehicle to the exact location where you want NPC vehicle to spawn, record the EGO vehicle's transform (position + rotation) information. And use this information to spawn the NPC vehicles.

Ego vehicle's transform information is not changing when we move the vehicle.

asifsid-32, 2020-08-20:
> A relatively simple solution is that you could drive the EGO vehicle to the exact location where you want NPC vehicle to spawn, record the EGO vehicle's transform (position + rotation) information. And use this information to spawn the NPC vehicles.

If you can provide me the script for ego vehicle's transform for every movement that will be great because i tried doing it i am not able to get information for every frame instead just for the first spawn. 

Please help !!

EricBoiseLGSVL, 2020-08-21:
@asifsid-32 
- Clone the SF map to Assets/ External/Environments/SanFrancisco/
- Open in Unity Editor
- Place the meshes (not the prefabs) in the scene
- Add colliders to the meshes in scene
- Save scene
- Build with Simulator -> Build...

Now you will have a new version on SF map with vehicles in the parking lot.
- You can also move the spawn Info object in scene to a place closer to the parking lot so you can load in near it.

For the transform data, you can make a new sensor or edit an existing sensor to output the transform data.  This could be sent via bridge or just added to the OnVisualize method in sensors.  Then make a new binary with the new or edited sensor

asifsid-32, 2020-08-21:
> @asifsid-32
>
> * Clone the SF map to Assets/ External/Environments/SanFrancisco/
> * Open in Unity Editor
> * Place the meshes (not the prefabs) in the scene
> * Add colliders to the meshes in scene
> * Save [...]

Eric.. The solutions you told by using or editing sensors, i am not able to find OnVisualize methods in sensors. Please help me with the json script that will be great help.

EricBoiseLGSVL, 2020-08-21:
KeyboardControlSensor.cs line 191 OnVisualize(Visualizer visualizer) is one example
All sensors have this method.  Here you can place graphData Dictionary transform data (or anything else you need).  Be sure to get the transform data in FixedUpdate or Update not Start.  When you visualize a sensor in the UI this data will populate.
If you want Bridge to send the data, you need a new sensor and set topic.
It is not JSON but the source code. e.g. C#
var transformPos = transform.position;
var transformRot = transform.rotation;

asifsid-32, 2020-08-21:
> KeyboardControlSensor.cs line 191 OnVisualize(Visualizer visualizer) is one example
> All sensors have this method. Here you can place graphData Dictionary transform data (or anything else you need). [...]

can i add this directly in json vehicle script

EricBoiseLGSVL, 2020-08-21:
Again, this is not JSON.  This is simulator source code C#.  Are you taking about the same thing?  You need to create a variable and update the transform data in the sensor code, then put this variable in the visualizer data.  Lastly, build a new sim binary so you can run the updated/changed code.

asifsid-32, 2020-08-21:
> Again, this is not JSON. This is simulator source code C#. Are you taking about the same thing? You need to create a variable and update the transform data in the sensor code, then put this variable in the visualizer data. Lastly, build a new sim binary so you can run the updated/changed code.

Eric can i get any link to follow the above mentioned steps ?

EricBoiseLGSVL, 2020-08-21:
https://docs.unity3d.com/ScriptReference/MonoBehaviour.Update.html

EricBoiseLGSVL, 2020-08-21:
@asifsid-32 Please post what code you have and I'll see if I can help with that.  Please use formatting like so.
```
var like = new this();
```

asifsid-32, 2020-08-21:
> @asifsid-32 Please post what code you have and I'll see if I can help with that. Please use formatting like so.
> 
> ```
> var like = new this();
> ```

@EricBoiseLGSVL  i am using create_npc python api. I need to spawn NPC at a particular place. Now as @ntutangyun suggested that was a great idea. But i am not able to get the transform data. Now as you said that is totally different code for me c# and again buildinf the sim. I need help in this .

EricBoiseLGSVL, 2020-08-21:
Do you have the simulator source code cloned or are you just using the binary?

asifsid-32, 2020-08-21:
> Do you have the simulator source code cloned or are you just using the binary?

Just using the executable file along with python api's

EricBoiseLGSVL, 2020-08-21:
Ok I see the issue, I thought you were working with the source code.  I recommend downloading the source code to make changes that you need for development.  As a workaround you could do what @ntutangyun said but with the python api.  You can make a .py that just outputs the transform position and drive to the parking spot. Record the position and then you can spawn the NPC at that position(s).  Here is the code.

'''
import os
import lgsvl
import math
import random

sim = lgsvl.Simulator(os.environ.get("SIMULATOR_HOST", "127.0.0.1"), 8181)
if sim.current_scene == "AutonomouStuff":
  sim.reset()
else:
  sim.load("BorregasAve")

spawns = sim.get_spawn()

state = lgsvl.AgentState()
state.transform = spawns[0]
a = sim.add_agent("Jaguar2015XE", lgsvl.AgentType.EGO, state)

sx = spawns[0].position.x
sy = spawns[0].position.y
sz = spawns[0].position.z

while True:
  print (a.transform)
'''

asifsid-32, 2020-08-21:
> Ok I see the issue, I thought you were working with the source code. I recommend downloading the source code to make changes that you need for development. As a workaround you could do what @ntutangyun said but with the python api. You can make a .py that just outputs the transform position and drive to the parking spot. Record the position and then you can spawn the NPC(s) at that position(s)/rotation(s). Here is the code.
> 
> ```
> import os
> import lgsvl
> import math
> import random
> 
> sim = lgsvl.Simulator(os.environ.get("SIMULATOR_HOST", "127.0.0.1"), 8181)
> if sim.current_scene == "AutonomouStuff":
>   sim.reset()
> else:
>   sim.load("AutonomouStuff")
> 
> spawns = sim.get_spawn()
> 
> state = lgsvl.AgentState()
> state.transform = spawns[0]
> a = sim.add_agent("Jaguar2015XE", lgsvl.AgentType.EGO, state)
> 
> sx = spawns[0].position.x
> sy = spawns[0].position.y
> sz = spawns[0].position.z
> 
> while True:
>   print (a.transform)
> ```
> 
> I saw an error and I edited. Also, you can use whatever vehicle you have, just change the EGO name

Facing Following errors @EricBoiseLGSVL ->

import: not authorized `os' @ error/constitute.c/WriteImage/1028.
import: not authorized `lgsvl' @ error/constitute.c/WriteImage/1028.
import: not authorized `math' @ error/constitute.c/WriteImage/1028.
import: not authorized `random' @ error/constitute.c/WriteImage/1028.
./quickstart/My_api.py: line 6: syntax error near unexpected token `('
./quickstart/My_api.py: line 6: `sim = lgsvl.Simulator(os.environ.get("SIMULATOR_HOST", "127.0.0.1"), 8181)'

asifsid-32, 2020-08-21:
> Ok I see the issue, I thought you were working with the source code. I recommend downloading the source code to make changes that you need for development. As a workaround you could do what [...]

Hii @EricBoiseLGSVL  i am able to spawn one NPC according to my preferred location. Now if i have to spawn multiple multiple NPc's can you suggest me the changes in the code ?

import os
import lgsvl
import random

sim = lgsvl.Simulator(os.environ.get("SIMULATOR_HOST", "127.0.0.1"), 8181)
if sim.current_scene == "BorregasAve":
  sim.reset()
else:
  sim.load("BorregasAve")

spawns = sim.get_spawn()

state = lgsvl.AgentState()
state.transform = spawns[0]
a = sim.add_agent("Lincoln2017MKZ (Apollo 5.0)", lgsvl.AgentType.EGO, state)

forward = lgsvl.utils.transform_to_forward(spawns[0])
right = lgsvl.utils.transform_to_right(spawns[0])

# Spawns one of each of the listed types of NPCS
# The first will be created in front of the EGO and then they will be created to the left
# The available types of NPCs can be found in NPCManager prefab
for i, name in enumerate(["Sedan"]):
  state = lgsvl.AgentState()

  # Spawn NPC vehicles 10 meters ahead of the EGO
  state.transform.position =  lgsvl.Vector(-75.8117370605469, -2.71694254875183, -46.0500717163086) 
  state.transform.rotation =  lgsvl.Vector(1.88953959941864, 232.987762451172, 359.851165771484)
  sim.add_agent(name, lgsvl.AgentType.NPC, state)

closed 2020-08-25 by asifsid-32