lgsvl/simulator #924: some NPC behaviors questions

qwetqwe, 2020-08-20:
1. in 98-npc-behavior.py, I ues`behaviours = sim.available_npc_behaviours`, but only get 3 behaviors:
 0: NPCLaneFollowBehaviour
 1: NPCWaypointBehaviour
 2: NPCManualBehaviour
where is the NPCDrunkDriverBehaviour and NPCTrailerBehaviour mentioned in py?
2. How can I use these behaviors? It seems that NPCManualBehaviour can control the NPC by manual, how to do it

EricBoiseLGSVL, 2020-08-21:
@qwetqwe Manual behavior is just a stub.  It hasn't been implemented yet.  ExampleNPCs repo has the DrunkDriver class that needs to be built in Editor to be able to be used in 98-npc-behavior.py.  What version of Simulator are you using?  Did this repo get released to public yet @bshin-lge ?

qwetqwe, 2020-08-26:
Hi,@EricBoiseLGSVL, I used lgsvlsimulator-linux64-2020.05 release version.

EricBoiseLGSVL, 2020-08-26:
@qwetqwe Yes, sorry for the confusion, we have not released this yet but we will soon.

qwetqwe, 2020-08-26:
@EricBoiseLGSVL ,thanks for you reply.
Expecting...

closed 2020-08-26 by qwetqwe