lgsvl/simulator #929: Starting the simulator from command line, not using the webUI

amirtakouh, 2020-08-24:
Hi,

Is it possible to start the simulator only from command line? without the need of going through the webUI?

Thanks and regards

EricBoiseLGSVL, 2020-08-24:
@amirtakouh Is this what you are looking for? https://www.lgsvlsimulator.com/docs/config-and-cmd-line-params/

amirtakouh, 2020-08-25:
@EricBoiseLGSVL the command line params mentioned in the link above handle only part of the needed configuration. One would still need to go through the webUI to select the map/vehicle/simulation configuration and press play to start the simulator. While the webUI is a nice and clean human interface, it is not very practical as a machine interface. It complicates the process of having the simulator running on a remote server and scripting around it.
Is there a possibility to circumvent the webUI all together?

EricBoiseLGSVL, 2020-08-25:
@zelenkovsky Adding you for Cloud/Server questions

amirtakouh, 2020-08-31:
@zelenkovsky could you shed some light on this? :)

EricBoiseLGSVL, 2020-08-31:
@amirtakouh As per another issue from @zelenkovsky 
Requirement to register and maintain logged in status is intentional, we expect to provide more cloud based features to the simulator in very near future. It would include running cloud simulation at scale, content sharing and etc.

While running continuously improving open source version of the simulator locally will be always free, most of value added premium features will be delivered via the cloud.

Our current license does not prevent modifying the source code for debugging and development of your AD system (which we consider a non commercial activity).

amirtakouh, 2020-09-01:
Thank you for the response.
The command line parameters already cover --username and --password, so the user can start the simulator and login from the command line. My question was mainly about the other needed configuration that is not possible through command line, such as selecting the car/map/simulation and pressing the play button. If there are any plans to accommodate these in the command line parameters.

EricBoiseLGSVL, 2020-09-01:
@amirtakouh We have a huge release planned after the next release.  This functionality is in development now and will release after.

closed 2020-10-20 by EricBoiseLGSVL