lgsvl/simulator #930: Us of LGSVL without ROS and Autoware

asifsid-32, 2020-08-24:
Hello LGSVL team, Can we totally use python api's to run simulator without any ROS and Autoware dependency ? If i want to connect simulator to my framework which is totally NON ROS can that be done ? Is it possible ? 

@EricBoiseLGSVL  can i make a python API combining all the other API's and use it without having any dependency of ROS and autoware and Apollo 5.0 ?

EricBoiseLGSVL, 2020-08-24:
@asifsid-32 What non ROS framework are you planning to use?  You can create a new bridge type in simulator source or wait until we release custom bridge types later this year.  I'm not sure what you are asking, "combining all other API's".  What other APIs?  ROS, Apollo?

hadiTab, 2020-08-24:
@asifsid-32 while it is possible to use the Python API I think the better solution is to develop a bridge for your custom framework to communicate with the simulator thruogh websocket. The upcoming release adds support for bridge-plugins which will make this easier.

If you want to use the Python API you would need to make an API call each time any of your sensors wants to publish a measurement. This would also slow down the simulator a lot since each API call pauses the simulator for one frame. Your AD stack would also need to be able to handle non-realtime situations. If you were to add realistic sensors that publish at high rates this would make the simulator run an order of magnitude slower than realtime.

asifsid-32, 2020-08-25:
> @asifsid-32 What non ROS framework are you planning to use?  You can create a new bridge type in simulator source or wait until we release custom bridge types later this year.  I'm not sure what you are asking, "combining all other API's".  What other APIs?  ROS, Apollo?

@EricBoiseLGSVL what I meant was I just want to pass simulator's data to my framework which is Non-Ros. Is it possible ??

EricBoiseLGSVL, 2020-08-25:
@asifsid-32 Ah, I see.  Yep, @hadiTab has the solution.  You can use websocket (which you can see how we built cyber and ROS) or even use UDP, which we used for a UDP Lidar sensor.  It's not easy but it's possible.  I'd wait until you can see our bridge plugin support releasing soon for guidance.

asifsid-32, 2020-08-25:
> @asifsid-32 Ah, I see.  Yep, @hadiTab has the solution.  You can use websocket (which you can see how we built cyber and ROS) or even use UDP, which we used for a UDP Lidar sensor.  It's not easy but it's possible.  I'd wait until you can see our bridge plugin support releasing soon for guidance.

@hadiTab can you explain the solution a little bit more ??

hadiTab, 2020-08-26:
@asifsid-32 I'm assuming your custom framework has some sort of mechanism for passing messages such as sensor data around. To communicate with the simulator you would need to send this data over websocket to the simulator. For example, for ROS, there are two "bridge" components. One runs as a ROS node, the other is part of the simulator. These two bridge components communicate with eachother over websocket. You would need to implement something similar for your custom framework. To make it easier to implement the simulator side without needing to rebuild the entire simulator we will be adding bridge plugin functionality in the next release which will allow you to include the simulator component of the bridge as a plugin that is loaded into the simulator on startup.

asifsid-32, 2020-09-03:
> @asifsid-32 I'm assuming your custom framework has some sort of mechanism for passing messages such as sensor data around. To communicate with the simulator you would need to send this data over [...]

Hii @hadiTab as the new release is out with bridge plugin functionality can you please help me with the steps that will help me as a communication mechanism between my framework and the simulator ?

Also @EricBoiseLGSVL  if I am not using Python API's since they make simulator too slow what can be done to directly run simulator and get the sensor data from simulator and pass it to my framework. Please help me with the steps as this seems too complicated to work on .

EricBoiseLGSVL, 2020-09-03:
@asifsid-32 Yes, the new custom bridge has been released.  We have documentation [here](https://www.lgsvlsimulator.com/docs/bridge-plugins/).
Please read the documentation and post issues that you have with the this.  We do have a small bug that we will be releasing a fix for, @hadiTab can confirm

asifsid-32, 2020-09-03:
> @asifsid-32 Yes, the new custom bridge has been released.  We have documentation [here](https://www.lgsvlsimulator.com/docs/bridge-plugins/).
> Please read the documentation and post issues that you have with the this.  We do have a small bug that we will be releasing a fix for, @hadiTab can confirm

@EricBoiseLGSVL this custom bridge won't be dependent on ROS and can be used simply with our framework ??

hadiTab, 2020-09-03:
@asifsid-32 yes the bridge does not depend on ROS. Also, the bug @EricBoiseLGSVL is referring to was actually fixed before the release so there should be no bugs right now.

EricBoiseLGSVL, 2020-09-03:
Awesome, thanks @hadiTab