lgsvl/simulator #931: Lane width calculation difference

yumianhuli2, 2020-08-24:
I used：
var ds = s - sOffset;
var widthValue = width.a + width.b * ds + width.c * ds * ds + width.d * ds * ds * ds;
var normalDir = GetNormalDir(curLeftBoundaryPoints, idx, isLeft);
var newPoint = point + (float)widthValue * normalDir;

To calculate the width of the reference line corresponding to lane
And I got this:
[image]
The point where 3lane becomes 2lane is
[image]

I also compared it with the odrViewer：
[image]
But the point where 3lane becomes 2lane is (40.5,-7.2)

So is there a bug in the lgsvl source formula or is there a problem with my usage?

The xodr file is here ：
[image]
plz change the file suffix name to xodr

luqiang21, 2020-08-24:
@yumianhuli2 There was a bug in computing normal directions. We should always use reference line points not the nearest left boundary points. That might be the reason for this issue. We have fixed this internally and it will go public very soon.

yumianhuli2, 2020-08-25:
> @yumianhuli2 There was a bug in computing normal directions. We should always use reference line points not the nearest left boundary points. That might be the reason for this issue. We have fixed this internally and it will go public very soon.

OK！I will test it in next release soon!:)

closed 2020-08-25 by yumianhuli2

yumianhuli2, 2020-08-31:
> @yumianhuli2 There was a bug in computing normal directions. We should always use reference line points not the nearest left boundary points. That might be the reason for this issue. We have fixed this internally and it will go public very soon.

I tried to always use the reference lines to calculate the normal direction, and then accumulated the widthValue last time, and the result was the same.Anyway, looking forward to the update and fix it these days！