lgsvl/simulator #932: Custom Traffic Light and Signs

gizegizem, 2020-08-24:
Hi , I am trying to use my own Traffic Light and Sign system .  I just followed the guide in the website. I have managed to do all traffic light change and other stuff  but npc's not obeying the traffic light. They just ran over them . 

I am trying to find the missing point . Thanks.

EricBoiseLGSVL, 2020-08-24:
@gizegizem Be sure that the traffic signals are annotated correctly and stoplines are referenced.  Please see CubeTown repo to see a simple map annotation done correctly.

gizegizem, 2020-08-26:
Thank you for answer @EricBoiseLGSVL as you said, I followed exact structure in CubeTown but as far as I see, the NPCs can do right turn after a second eventhough the light is red. But if the car tries to do left turn or go straight , it just waits and obey the traffic light. How can I manage to npc's not to turn right immediately . They just wait 1-2 sec and turn right.

gizegizem, 2020-08-26:
By the way , I also try to add speed bump however I couldn't find AutonomousStaff Speed Bump Mesh. How can I find it ? Or is there any solution

alperenbayraktar, 2020-08-26:
I was dealing with the same problem earlier today, traced back the code, and saw that the MapIntersection.cs initializes the TrafficLightLoop. I had my traffic lights on the MapLaneSections, not in the MapIntersections. Also had a TrafficLight.cs but decided to delete my code and assets and just copy the traffic lights in CubeTown and thus, their code. 
- Carried my MapSignals into according MapIntersection objects.
* Did not work, then checked https://www.lgsvlsimulator.com/docs/map-annotation/#create-stop-lines , your stop lines too should be in MapIntersections. 
- Carried my StopLines into according MapIntersection objects.
[image]
* Cars used to stop when the stop line indicates "Red" but, after a quick while they continue to move.
- Aligned my stop lines a bit far from MapSignal because the front section of the cars was passing the traffic light so they continued.
[image]

Now, it works perfectly. Hope this helps!

EricBoiseLGSVL, 2020-08-26:
@alperenbayraktar Thanks for the helpful post!