lgsvl/simulator #933: A few questions about working with the simulator

dymons, 2020-08-24:
Hello! While I was studying the simulator, I had some questions that I couldn't find an answer to in the documentation.
Could you explain to me some things related to the simulator (I don't know Unity).

1. Can I set my own time factor to deceleration/acceleration time? For example, the Carla simulator [Setting a time factor](https://carla.readthedocs.io/en/0.9.8/adv_recorder/).
2. Allow traffic participants to break the rules.
3. Creating uneven surfaces.
4. Changing material characteristics (for example, for a road).
5. Running multiple EGO cars in the simulator with my AD stack.
6. Rewind time during simulation.

In advance, thank you very much for the explanation.

lemketron, 2020-08-24:
LGSVL Simulator supports multiple ego vehicles. Refer to [Maps & Vehicles](https://www.lgsvlsimulator.com/docs/simulations-tab/#how-to-addedit-a-simulation) documentation, specifically `Bridge Connection String` and `+` button for adding additional ego vehicles to a simulation. Each ego will need to be controlled by a separate instance of a bridge to the AD software that will control it.

Refer to [Non-realtime simulation](https://www.lgsvlsimulator.com/docs/python-api/#non-realtime-simulation) documentation for information on setting the time factor.

I'll let @EricBoiseLGSVL respond to the other questions.

EricBoiseLGSVL, 2020-08-24:
@dymons Hello and welcome to LGSVL Simulator
1. Yes, thanks @lemketron 
2. Yes, you will need to edit existing NPC and Pedestrian behavior classes.  We will soon release features to make this easier but you can edit as you need.
3. Yes, you can create any map that you need.  You just need a modeling tool like Maya or Blender.
4. I assume you mean physics materials and if so yes.  You can create a new physics material and apply to a mesh
5. Yes, thanks @lemketron 
6. No but we are working on being able to do this with a visual scenario editor and this will release later this year.

Since LGSVL Simulator is open-source, you can make any changes you need and create a new binary.  Please be sure to enable GitLFS when cloning our open environments and vehicles.

dymons, 2020-08-25:
@lemketron, @EricBoiseLGSVL Thank you for your explanation!

closed 2020-08-25 by dymons