lgsvl/simulator #934: Using ROS with pythonAPI

Maddy2927, 2020-08-24:
I was trying to incorporate ROS publisher and subscriber to get the different sensor data using python API. Is it possible and a viable solution?

UYasher, 2020-08-24:
You can get that data via the python API, as described [here](https://www.lgsvlsimulator.com/docs/python-api/#sensors). You can also look at the [examples](https://www.lgsvlsimulator.com/docs/api-quickstart-descriptions/) to get a clearer idea of how to implement this.

However, depending on what you are doing, the python API might not be the best option. Every time you access the API, the simulation pauses. Depending on what you want to do, this may be fine. If you want to get the data while running the simulation, you should be using ROS, Autoware, or Apollo though.

(Members of the LGSVL team might have more to say as well, this is just from my person experience using the simulator.)

EricBoiseLGSVL, 2020-08-24:
@Maddy2927 Yes, it is best to use the bridge for ROS messages, not API.  You can create your own bridge type in simulator or you can wait until we release support for custom bridge types later this year.

Maddy2927, 2020-08-25:
@UYasher  @EricBoiseLGSVL  Thanks for the response. Can you in brief explain me how to use bridge concept to use my ROS dependent codes in the simulator. 
Do we have to make use of dockers and containers for that as mentioned in the documentation or is their any other way we can use bridge without these concepts?

EricBoiseLGSVL, 2020-08-25:
@Maddy2927 Sure.  Please look at the current sensors we have in the project.  You can see how GPSSensor.cs adds a publisher and queues the data to be sent via the topic.  On the flip side you can see how VehicleControlSensor.cs adds a subscriber to the topic to receive and parse the control data.  Bridge does not require docker but you will need to alter source code to add your own sensor to suit your needs.  Hope this helps.

Maddy2927, 2020-09-04:
@EricBoiseLGSVL  Also codes which i want to see the output of in the simulator are all written in C++ with ROS as a wrapper. Is it possible for me to run those codes directly in the simulator. Are c++ API supported by the simulator like python?

EricBoiseLGSVL, 2020-09-04:
@Maddy2927 You can create a dll of the c++ and place in simulator.  We also have custom bridge functionality that use ROS.  Please check the latest release documentation to see if that helps