lgsvl/simulator #935: Path Not Executing With Two Computer Setup (LGSVL and Apollo)

UYasher, 2020-08-24:
I currently have two computers, one running LGSVL and the other running Apollo, connected by a gigabit ethernet connection. Apollo is able to receive data from LGSVL effectively in real-time. Things seems to work until the path planning step, and the message "Path planned successfully!" appears after setting the path. However, the car does not move after that, and the blue lines on the road in Apollo which show the path remaining never appear (although the red line does). This was not the case when I ran both LGSVL and Apollo on the same machine. At that time, the car executed the planned path. I switched to two computers because there was significant lag.

Any idea what the issue might be or how to resolve it? Help appreciated.

rongguodong, 2020-08-24:
Can you post a screenshot of cyber_monitor? Do you have all these channels as green as shown in this picture?
[image]

rongguodong, 2020-08-24:
Which version of Apollo are you using?

UYasher, 2020-08-24:
I'm using the latest Apollo pre-6. Attached is a screenshot with the cyber_monitor output:

[image]

rongguodong, 2020-08-24:
Can you check the content of "apollo/planning" channel?

UYasher, 2020-08-24:
[image]

rongguodong, 2020-08-24:
That is the reason. Planning is not ready.
You image channels have too high frequency, which indicates you may have enabled "Camera" module. Can you try it again by just enable "Localization, Transform, Perception, Traffic Light, Planning, Prediction, Routing, and Control" modules? Not sure if it is related though.

rongguodong, 2020-08-24:
How do you set the destination? Do you set it manually in "Route Editing" tab?

UYasher, 2020-08-24:
> How do you set the destination? Do you set it manually in "Route Editing" tab?
Yes

> Can you try it again by just enable "Localization, Transform, Perception, Traffic Light, Planning, Prediction, Routing, and Control" modules? Not sure if it is related though.
Made that change. Here is the /apollo/planning channel
[image]

rongguodong, 2020-08-24:
Planning is still failing. Can you check planning's log under /data/log to see if there is anything suspicious?

UYasher, 2020-08-24:
It repeatedly shows --
E0824 14:04:21.510143  8344 reference_line_provider.cc:196] Routing is not ready.

Followed by these three lines repeated for a while --
E0824 14:04:30.472756  8342 planning_component.cc:153] Input check failed
E0824 14:04:30.556586  8348 reference_line_provider.cc:196] Routing is not ready.
E0824 14:04:30.556668  8346 planning_component.cc:225] routing not ready; skip the planning cycle.

Followed by these 3 repeating after it receives the routing data --
E0824 14:04:34.561844  8348 reference_line_provider.cc:203] Fail to get reference line
E0824 14:04:34.648946  8341 reference_line_provider.cc:538] Failed to extract segments from routing
E0824 14:04:34.648952  8341 reference_line_provider.cc:580] Failed to create reference line from routing

In prior logs, I've also seen occasional buffer overflows like the one below, although I didn't catch any in the file --

W0817 17:08:15.526425 32903 channel_buffer.h:70] channel[/apollo/prediction] read buffer overflow, drop_message[1] pre_index[399] current_index[400]

In so far as I can tell, I think that's everything in the log file?

rongguodong, 2020-08-24:
It seems routing module has problem? Can you also check the log of routing?

Another thing: You said when you run simulator and Apollo on same machine, there is no problem. The problem only happens when you run them on different machine, right? This is very strange since it should not matter where the simulator is run. Can you try both situation and find out what difference is there. Do you run exactly same simulation with same sensor config (JSON)? Do you set the same destination? And difference of cyber_monitor result of both situations (running on same machine vs. running on different machines)?

rongguodong, 2020-08-24:
Can you post these three log files here: planning, routing, map?

UYasher, 2020-08-24:
> Can you post these three log files here: planning, routing, map?

Here are the log files.
[routing.log](https://github.com/lgsvl/simulator/files/5119785/routing.log)

> Do you set the same destination? 

Yes

I am also checking the other points you mentioned.

[map.log](https://github.com/lgsvl/simulator/files/5119782/map.log)
[planning.log](https://github.com/lgsvl/simulator/files/5119783/planning.log)

rongguodong, 2020-08-24:
It failed in [this function](https://github.com/ApolloAuto/apollo/blob/pre6/modules/map/pnc_map/pnc_map.cc#L416). Please search the error messages of this function (e.g. line 421, 428, 441, 451, 457, 466, etc.) to find which log(s) contains such messages. These messages should tell us why this function failed.

UYasher, 2020-08-25:
I'm not certain what exactly fixed the issue, but it's working like a charm now. Thanks for the help @rongguodong

rongguodong, 2020-08-25:
I will close this issue for now.

If anyone met the same issue, please reopen this. We really want to figure out the root cause of this. If you see the same issue, please search the error messages from PncMap::GetRouteSegments function (see my comment above for more details). This would help us to understand why this function can fail.

closed 2020-08-25 by rongguodong