lgsvl/simulator #936: Adding New Vehicle Invalid: End of stream

amc-nu, 2020-08-25:
After following the tutorial from:
https://www.youtube.com/watch?v=R_Z65kimCII

1. New Vehicle AssetBundle is shown as correctly loaded in the WebUI
2. While pressing the play button to launch a new simulation I get the `Invalid: End of Stream` error
3. The simulation doesn't start.

**Unity Version:** 2019.3.3f1
**Simulator Version:** lgsvlsimulator-windows64-2020.03

Thanks

EricBoiseLGSVL, 2020-08-25:
@amc-nu Please post your player.log so we can see better what is going on.  Also, any images of the vehicle setup in unity will help.  Thanks for the versions data, it helps.

amc-nu, 2020-08-27:
Thank you for your prompt reply.
Apologies for my late reply @EricBoiseLGSVL 
I was able to solve this problem recompiling the simulator binary from inside the Unity editor. 
Simulator -> Build -> Build Simulator.

closed 2020-08-27 by amc-nu

EricBoiseLGSVL, 2020-08-27:
@amc-nu No worries, glad you were able to solve the problem and thanks for posting your fix to help others.