lgsvl/simulator #937: Hardware requirements for running LGSVL simulator smoothly

yumianhuli2, 2020-08-25:
Hi! Lgsvl Team!
    CPU：Intel Xeon Gold 6136 or higher
    memory：64 GB
    GPU：NVIDIA GeForce GTX 1080Ti/2080Ti/Titan

Can such a configuration run LGSVL simulator and Apollo5.5 algorithms smoothly on one PC now？
THANKS!

lemketron, 2020-08-25:
The Ti variants have more GPU memory so there’s a chance everything might fit. 

Performance-wise you should be able to run Apollo 5.0 on the same machine with the simulator (using a 1080, as the 2080 is not supported by 5.0).

But from what we’ve seen so far, 5.5 with camera and LiDAR perception (and traffic light) modules will be pretty demanding. It’s too soon to tell, though, since camera perception (and thus also traffic light) modules are not working yet in Apollo master.

yumianhuli2, 2020-08-26:
> The Ti variants have more GPU memory so there’s a chance everything might fit.
>
> Performance-wise you should be able to run Apollo 5.0 on the same machine with the simulator (using a 1080, as the [...]

Thank you!

closed 2020-08-26 by yumianhuli2