lgsvl/simulator #938: Save Camera_Output in the form of Video

asifsid-32, 2020-08-25:
Hello LGSVL team,

Can we use save_camera_image python api to generate videos captured from the main camera and other cameras if added to json scripts ?

EricBoiseLGSVL, 2020-08-25:
@asifsid-32 Yes, what ever encoding libraries you want can be used to create videos from images.  We already have this and plan on releasing in the near future.  You can implement in simulator or outside after all images are captured.

asifsid-32, 2020-08-25:
> @asifsid-32 Yes, what ever encoding libraries you want can be used to create videos from images.  We already have this and plan on releasing in the near future.  You can implement in simulator or outside after all images are captured.

@EricBoiseLGSVL can you please help me the API here ?? I tried using the given api but all I was able to do was getting the image dumped of the last frame !!

EricBoiseLGSVL, 2020-08-25:
Are you looping to continuously capture the images in the .py?

asifsid-32, 2020-08-25:
> Are you looping to continuously capture the images in the .py?

@EricBoiseLGSVL yes I tried that !!

asifsid-32, 2020-08-25:
> Are you looping to continuously capture the images in the .py?

As you mentioned about the encoding libraries, I tried with .avi but of no help !!

EricBoiseLGSVL, 2020-08-25:
@asifsid-32 I'm confused, are you having issues with capturing images from the API or encoding?

asifsid-32, 2020-08-25:
> @asifsid-32 I'm confused, are you having issues with capturing images from the API or encoding?

@EricBoiseLGSVL what I basically need is the output from my main camera and other cameras which I haved added to my ego vehicle in the form of video. Now I tried saving video the same way you have mentioned in the .py file using save(). 

I wanted to ask how can I get the video output from my main camera ??

EricBoiseLGSVL, 2020-08-25:
@asifsid-32 To save video you will need to use an encoder.  I recommend using ffmpeg which we used to encode directly in simulator as the images are captured from the camera rendertexture.  You can also use a simple approach and use the encoder after you have collected all the images with the API.  [Here](https://ostechnix.com/20-ffmpeg-commands-beginners/) is a site that will help.

asifsid-32, 2020-08-25:
> @asifsid-32 To save video you will need to use an encoder.  I recommend using ffmpeg which we used to encode directly in simulator as the images are captured from the camera rendertexture.  You can [...]

@EricBoiseLGSVL Also I facing this problem where I am getting only the last frame image from the simulator dumped to my system. Not able to get all the images in between the starting frame and the ending frame !!!

asifsid-32, 2020-08-26:
> Are you looping the API save method in the .py file or are you running it once?

Looping the API save method in .py

asifsid-32, 2020-08-26:
> Are you looping the API save method in the .py file or are you running it once?

Got it @EricBoiseLGSVL

EricBoiseLGSVL, 2020-08-26:
Can you please post the error? Also, can you post the relevant code that you are using?

asifsid-32, 2020-08-27:
> Can you please post the error? Also, can you post the relevant code that you are using?

Hello @EricBoiseLGSVL I was able to solve the error !! Thank you

EricBoiseLGSVL, 2020-08-27:
@asifsid-32 Excellent!  Please post code or a description of the solution so other users that might have the same issue can find it.  Thanks.

asifsid-32, 2020-08-27:
> @asifsid-32 Excellent!  Please post code or a description of the solution so other users that might have the same issue can find it.  Thanks.

Sure !! I will .

OrbitXN, 2021-11-07:
@asifsid-32 can you post your solution? I have the same problem...