lgsvl/simulator #940: Register inputs/key press even if pressed for a short period

dr563105, 2020-08-25:
Hello,
Currently when I press the up arrow to accelerate, in the keyboard sensor HUD, acceleration turns 1 only when the duration of the key press is ca.0.5s. Consecutive repeated presses reflect the velocity to raise but not acceleration's discrete value. Same goes with braking(down arrow). However, the steering is fine. It registers the turns. nudges easily. How can I achieve this for acceleration and braking? 
I ask this because, once accelerated, the control remains `unchanged` till I long press again. During data collection phase this behaviour generates imbalanced classes. So is there a way to register the key strokes even for minute presses? 

For example, I tried changing this line https://github.com/lgsvl/LaneFollowingSensor/blob/master/LaneFollowingSensor.cs#L68 and gave a much smaller value. I didn't see an improvement. Would changing this line https://github.com/lgsvl/simulator/blob/release-2020.05/Assets/Scripts/Dynamics/FMU/IVehicleDynamics.cs#L57 ?

Thank you.

EricBoiseLGSVL, 2020-08-26:
@dr563105 Steering is lerped to give gradual rotation to the wheels.  You will need to do the same for acceleration.  We did not do this because we wanted AD stack to provide raw input.  You should look at KeyboardControlSensor.cs to have this gradual acceleration instead of raw keyboard input

dr563105, 2020-08-26:
Thanks Eric. I see it use Mathf.Movetowards(). I’d like the acceleration and braking to be just discrete values— `-1 0 1` but changing gradually with time. Can you help me figure out which math function in unity will achieve this?

EricBoiseLGSVL, 2020-08-26:
No prob.  I think if you want discrete you can check value, e.g. inputValue > 0 then 1 else < 0 then -1 else 0.  All apis for this is for incremental values not discrete.  I think this is an issue with Unity's input system. We have an older version that we need to update.

dr563105, 2020-08-31:
I'm closing this for now. Will reopen if needed.

closed 2020-08-31 by dr563105