lgsvl/simulator #942: lgsvl_msgs: Viewing Traffic light bounding boxes

ashwinsarvesh, 2020-08-26:
Hello,

I am trying to view the bounding boxes for traffic light signals by subscribing to the ground truth data
I am using LGSVL fork of Apollo 5.0
But my _Apollo.5.0/lgsvl_pkgs/lgsvl_msgs_ folder is empty.
Can someone tell me what should I do next ?

dr563105, 2020-08-26:
Have you already tried to follow these instructions? -- https://www.lgsvlsimulator.com/docs/lgsvl-msgs/

ashwinsarvesh, 2020-08-26:
@dr563105 This can be also used for Apollo 5.0 which uses the CyberRT bridge?
I see only ROS/ROS2 mentioned in the link.

rongguodong, 2020-08-26:
@ashwinsarvesh How do you clone our fork of Apollo 5.0? You should get proto files in lgsvl_msgs. It is a submodule pointing to another repo ("proto" branch). If you cannot get it for some reason, you can manually clone them from here:
https://github.com/lgsvl/lgsvl_msgs/tree/protobuf

rongguodong, 2020-08-26:
If you follow our instruction [here](https://www.lgsvlsimulator.com/docs/apollo5-0-instructions/#cloning-the-repository), you should automatically get its contents in "lgsvl_msgs".

closed 2020-09-03 by EricBoiseLGSVL

ashwinsarvesh, 2020-09-04:
Forgot to reply.
Thank you all for your answers

EricBoiseLGSVL, 2020-09-04:
@ashwinsarvesh no worries, glad to help :)