lgsvl/simulator #943: Question about import pointCloud and export Opendrive format map.

wisdom-bob, 2020-08-26:
Now I can import .pcd file into the simulator. However, I find some question: 
1. my .pcd file with rgb data can't be imported successfully. Can I ask what the format API database you are referring to? Or can you give me more information on how to use the point cloud import module? as below is the format I find in the project. I'm not familiar with C#. I feel that the data structure is too casual. I don't know which data structure are feasible:
namespace Simulator.Editor.PointCloud
{
    using System;

    public enum PointElementType
    {
        Byte,
        Float,
        Double,
    }

    public enum PointElementName
    {
        X, Y, Z,
        R, G, B,
        I,
    }
    
    public struct PointElement
    {
        public PointElementType Type;
        public PointElementName Name;
        public int Offset;

        public static int GetSize(PointElementType type)
        {
            switch (type)
            {
                case PointElementType.Byte: return 1;
                case PointElementType.Float: return 4;
                case PointElementType.Double: return 8;
            }
            throw new Exception($"Unsupported point element type {type}");
        }
    }
}
2. I think there is a size limit for PCD files that support importing. Is there any way to increase this limit? When I want to import a larger file, the following error will appear.
UnauthorizedAccessException: Access to the path is denied.
System.IO.MemoryMappedFiles.MemoryMapImpl.Map (System.IntPtr handle, System.Int64 offset, System.Int64& size, System.IO.MemoryMappedFiles.MemoryMappedFileAccess access, System.IntPtr& mmap_handle, System.IntPtr& base_address) (at <351e49e2a5bf4fd6beabb458ce2255f3>:0)
System.IO.MemoryMappedFiles.MemoryMappedView.Create (System.IntPtr handle, System.Int64 offset, System.Int64 size, System.IO.MemoryMappedFiles.MemoryMappedFileAccess access) (at <351e49e2a5bf4fd6beabb458ce2255f3>:0)
System.IO.MemoryMappedFiles.MemoryMappedFile.CreateViewAccessor (System.Int64 offset, System.Int64 size, System.IO.MemoryMappedFiles.MemoryMappedFileAccess access) (at <351e49e2a5bf4fd6beabb458ce2255f3>:0)
Simulator.Editor.PointCloud.Trees.PointProcessor.CalculateBoundsDefault .......

here is a question about export the map in Opendrive v1.4 format.
1. as I know, Opendrive use polynomial to express curve. However, the exported map, as i see, uses many straight lines to express curve. Can I try to use polynomial?

EricBoiseLGSVL, 2020-08-26:
@wisdom-bob We just implemented polynomial for open drive and should be available soon.  I'll add @luqiang21 to confirm
@kamkolak Offered help in issue #921.  Is this the same issue or different?

wisdom-bob, 2020-08-27:
I'm looking forward to the new feature! How about the PointCloud import function? As I know,PCL is now the most popular pcd api. However, it seems not support LGSVL import function. So which api is the designed import function reference？

luqiang21, 2020-08-27:
@wisdom-bob Yes, we just implemented polynomial fitting within each reference line. Hope it will be helpful for you.

wisdom-bob, 2020-08-27:
> @wisdom-bob Yes, we just implemented polynomial fitting within each reference line. Hope it will be helpful for you.

But, as I see in the exported .xodr file in Opendrivev1.4 format, the curve line is cut into many straight lines. As below is the exported file according to the CubeTown Annotation data. Emm....... Sorry, I think I'm wrong. I think it should be expressed by a few polynomial curve segments. However, now it's too many many segments which i think it's not necessary. 

> 
<lane id="1" type="driving" level="false">
                        <link>
                            <predecessor id="-1" />
                            <successor id="1" />
                        </link>
                        <width sOffset="0" a="6.4771332740783691" b="0.077750742435455322" c="0" d="0" />
                        <width sOffset="0.9710274338722229" a="6.5526313781738281" b="0.10557156056165695" c="0.021658357232809067" d="-0.0072007821872830391" />
                        <width sOffset="1.9420549869537354" a="6.6689729690551758" b="0.11580397933721542" c="0.00068226415896788239" d="0.0049540190957486629" />
                        <width sOffset="2.9130327701568604" a="6.7865948677062988" b="0.18377673625946045" c="0.015052645467221737" d="-0.050936795771121979" />
                        <width sOffset="3.8829331398010254" a="6.932525634765625" b="-0.032658472657203674" c="-0.13328032195568085" d="0.057243622839450836" />
                        <width sOffset="4.8539609909057617" a="6.8275551795959473" b="-0.04480237141251564" c="0.033474724739789963" d="-0.032660137861967087" />
                        <width sOffset="5.8249878883361816" a="6.7857112884521484" b="-0.128217875957489" c="-0.061564624309539795" d="0.026879418641328812" />
                        <width sOffset="6.79429817199707" a="6.6280641555786133" b="-0.15596194565296173" c="0.016627557575702667" d="0.010048201307654381" />
                        <width sOffset="7.765324592590332" a="6.5014986991882324" b="-0.082464814186096191" c="0.045766737312078476" d="-0.0036446452140808105" />
                        <width sOffset="8.7266912460327148" a="6.461280345916748" b="0.0059775826521217823" c="0.035364288836717606" d="-0.014946640469133854" />
                        <width sOffset="9.6977176666259766" a="6.4867448806762695" b="0.0052696117199957371" c="-0.0079667260870337486" d="0.01401987299323082" />
                        <width sOffset="10.661507606506348" a="6.4969744682312012" b="0.037642620503902435" c="0.032637055963277817" d="0.0047651128843426704" />
                        <width sOffset="11.632534980773926" a="6.5686626434326172" b="0.17537172138690948" c="0.046517878770828247" d="-0.059788472950458527" />
                        <width sOffset="12.603562355041504" a="6.7280740737915039" b="0.0033225114457309246" c="-0.12657354772090912" d="0.040236398577690125" />
                        <width sOffset="13.563822746276855" a="6.6501784324646" b="-0.10211458802223206" c="-0.010357286781072617" d="0.011983329430222511" />
                        <width sOffset="14.534852027893066" a="6.5522284507751465" b="-0.070153877139091492" c="0.024551233276724815" d="-0.0072952494956552982" />
                        <width sOffset="15.505881309509277" a="6.5005769729614258" b="-0.052700940519571304" c="0.0033038628753274679" d="0.0028810568619519472" />
                        <width sOffset="16.476503372192383" a="6.45517110824585" b="-0.039047751575708389" c="0.01169231440871954" d="0.0038387994281947613" />
                        <width sOffset="18.418558120727539" a="6.4317941665649414" b="0.020350208505988121" c="0" d="0" />
                        <roadMark sOffset="0" type="solid" weight="standard" color="standard" laneChange="none" />
                    </lane>

yumianhuli2, 2020-08-27:
> @wisdom-bob Yes, we just implemented polynomial fitting within each reference line. Hope it will be helpful for you.

Hope to fix the Polyparam of referenceline  and the calculation of the width of laneWidth in the next version.By the way, when is the next version update expected：）？

luqiang21, 2020-08-27:
> > @wisdom-bob Yes, we just implemented polynomial fitting within each reference line. Hope it will be helpful for you.
> 
> But, as I see in the exported .xodr file in Opendrivev1.4 format, the curve line is cut into many straight lines. As below is the exported file according to the CubeTown Annotation data. Emm....... Sorry, I think I'm wrong that it's cut into many curve segments. I think it should be expressed by a few polynomial curve segments. However, now it's too many many segments which i think is not necessary.
> 
> > 
> 
> lane id="1" type="driving" level="false"
> 
> ```
> link
>     predecessor id="-1" /
>     successor id="1" /
> /link
> width sOffset="0" a="6.4771332740783691" b="0.077750742435455322" c="0" d="0" /
> width sOffset="0.9710274338722229" a="6.5526313781738281" b="0.10557156056165695" c="0.021658357232809067" d="-0.0072007821872830391" /
> width sOffset="1.9420549869537354" a="6.6689729690551758" b="0.11580397933721542" c="0.00068226415896788239" d="0.0049540190957486629" /
> width sOffset="2.9130327701568604" a="6.7865948677062988" b="0.18377673625946045" c="0.015052645467221737" d="-0.050936795771121979" /
> width sOffset="3.8829331398010254" a="6.932525634765625" b="-0.032658472657203674" c="-0.13328032195568085" d="0.057243622839450836" /
> width sOffset="4.8539609909057617" a="6.8275551795959473" b="-0.04480237141251564" c="0.033474724739789963" d="-0.032660137861967087" /
> width sOffset="5.8249878883361816" a="6.7857112884521484" b="-0.128217875957489" c="-0.061564624309539795" d="0.026879418641328812" /
> width sOffset="6.79429817199707" a="6.6280641555786133" b="-0.15596194565296173" c="0.016627557575702667" d="0.010048201307654381" /
> width sOffset="7.765324592590332" a="6.5014986991882324" b="-0.082464814186096191" c="0.045766737312078476" d="-0.0036446452140808105" /
> width sOffset="8.7266912460327148" a="6.461280345916748" b="0.0059775826521217823" c="0.035364288836717606" d="-0.014946640469133854" /
> width sOffset="9.6977176666259766" a="6.4867448806762695" b="0.0052696117199957371" c="-0.0079667260870337486" d="0.01401987299323082" /
> width sOffset="10.661507606506348" a="6.4969744682312012" b="0.037642620503902435" c="0.032637055963277817" d="0.0047651128843426704" /
> width sOffset="11.632534980773926" a="6.5686626434326172" b="0.17537172138690948" c="0.046517878770828247" d="-0.059788472950458527" /
> width sOffset="12.603562355041504" a="6.7280740737915039" b="0.0033225114457309246" c="-0.12657354772090912" d="0.040236398577690125" /
> width sOffset="13.563822746276855" a="6.6501784324646" b="-0.10211458802223206" c="-0.010357286781072617" d="0.011983329430222511" /
> width sOffset="14.534852027893066" a="6.5522284507751465" b="-0.070153877139091492" c="0.024551233276724815" d="-0.0072952494956552982" /
> width sOffset="15.505881309509277" a="6.5005769729614258" b="-0.052700940519571304" c="0.0033038628753274679" d="0.0028810568619519472" /
> width sOffset="16.476503372192383" a="6.45517110824585" b="-0.039047751575708389" c="0.01169231440871954" d="0.0038387994281947613" /
> width sOffset="18.418558120727539" a="6.4317941665649414" b="0.020350208505988121" c="0" d="0" /
> roadMark sOffset="0" type="solid" weight="standard" color="standard" laneChange="none" /
> ```
> 
> /lane

It is not released yet. And our map annotations are only discrete points, so we had to generate equations for every two points.

wisdom-bob, 2020-08-29:
> > > @ wisdom-bob是的，我们只是在每条参考线内实现了多项式拟合。希望对您有帮助。
> > 
> > 
> > 但是，正如我在以Opendrivev1.4格式导出的.xodr文件中看到的那样，曲线被切成许多直线。如下是根据CubeTown注释数据导出的文件。嗯.........对不起，我认为将其切成许多曲线段是错误的。我认为应该用几个多项式曲线段来表示。但是，现在有太多细分市场，我认为没有必要。
> > > 
> > 
> > 
> > 车道ID =“ 1” type =“ driving” level =“ false”
> > ```
> > link
> >     predecessor id="-1" /
> >     successor id="1" /
> > /link
> > width sOffset="0" a="6.4771332740783691" b="0.077750742435455322" c="0" d="0" /
> > width sOffset="0.9710274338722229" a="6.5526313781738281" b="0.10557156056165695" c="0.021658357232809067" d="-0.0072007821872830391" /
> > width sOffset="1.9420549869537354" a="6.6689729690551758" b="0.11580397933721542" c="0.00068226415896788239" d="0.0049540190957486629" /
> > width sOffset="2.9130327701568604" a="6.7865948677062988" b="0.18377673625946045" c="0.015052645467221737" d="-0.050936795771121979" /
> > width sOffset="3.8829331398010254" a="6.932525634765625" b="-0.032658472657203674" c="-0.13328032195568085" d="0.057243622839450836" /
> > width sOffset="4.8539609909057617" a="6.8275551795959473" b="-0.04480237141251564" c="0.033474724739789963" d="-0.032660137861967087" /
> > width sOffset="5.8249878883361816" a="6.7857112884521484" b="-0.128217875957489" c="-0.061564624309539795" d="0.026879418641328812" /
> > width sOffset="6.79429817199707" a="6.6280641555786133" b="-0.15596194565296173" c="0.016627557575702667" d="0.010048201307654381" /
> > width sOffset="7.765324592590332" a="6.5014986991882324" b="-0.082464814186096191" c="0.045766737312078476" d="-0.0036446452140808105" /
> > width sOffset="8.7266912460327148" a="6.461280345916748" b="0.0059775826521217823" c="0.035364288836717606" d="-0.014946640469133854" /
> > width sOffset="9.6977176666259766" a="6.4867448806762695" b="0.0052696117199957371" c="-0.0079667260870337486" d="0.01401987299323082" /
> > width sOffset="10.661507606506348" a="6.4969744682312012" b="0.037642620503902435" c="0.032637055963277817" d="0.0047651128843426704" /
> > width sOffset="11.632534980773926" a="6.5686626434326172" b="0.17537172138690948" c="0.046517878770828247" d="-0.059788472950458527" /
> > width sOffset="12.603562355041504" a="6.7280740737915039" b="0.0033225114457309246" c="-0.12657354772090912" d="0.040236398577690125" /
> > width sOffset="13.563822746276855" a="6.6501784324646" b="-0.10211458802223206" c="-0.010357286781072617" d="0.011983329430222511" /
> > width sOffset="14.534852027893066" a="6.5522284507751465" b="-0.070153877139091492" c="0.024551233276724815" d="-0.0072952494956552982" /
> > width sOffset="15.505881309509277" a="6.5005769729614258" b="-0.052700940519571304" c="0.0033038628753274679" d="0.0028810568619519472" /
> > width sOffset="16.476503372192383" a="6.45517110824585" b="-0.039047751575708389" c="0.01169231440871954" d="0.0038387994281947613" /
> > width sOffset="18.418558120727539" a="6.4317941665649414" b="0.020350208505988121" c="0" d="0" /
> > roadMark sOffset="0" type="solid" weight="standard" color="standard" laneChange="none" /
> > ```
> > 
> > 
> > /车道
> 
> 它尚未发布。而且我们的地图注释只是离散点，因此我们必须为每两个点生成方程式。

Sorry, But I have to say, there seem to be a bigger trouble. I export a map in Opendrive format. Finally, I find the curve in map is expressed by several straightline so that a strange phenomenon of pavement wrinkles appears. I think the curve application is urgent needed！

EricBoiseLGSVL, 2020-08-31:
@wisdom-bob Yes, we have included this in the next release

closed 2020-09-02 by wisdom-bob

harderthan, 2020-12-01:
> @wisdom-bob Yes, we have included this in the next release

Hi, @wisdom-bob. I also looking for the function, the PointCloud import function, what @wisdom-bob sugguested.
When will it be released specifically?

EricBoiseLGSVL, 2020-12-01:
We are putting together the release now, targeted for this month.

harderthan, 2021-01-20:
> We are putting together the release now, targeted for this month.

@EricBoiseLGSVL , Is there a update about the release? 
Could you share whether the schedule has been postponed or not?

EricBoiseLGSVL, 2021-01-22:
We have postponed until Feb.