lgsvl/simulator #945: Fish_eye Camera too bright

asifsid-32, 2020-08-27:
Hii LGSVL Team, i used fish_eye camera mounted on my ego car and the output is too bright to see the scenario. Can we do anything in the json file config to reduce the brightness ?
[image]

rongguodong, 2020-08-27:
It is related to auto exposure feature in HDRP. We are investigating the solution for that.

EricBoiseLGSVL, 2020-08-27:
In the meantime you can set Exposure to fixed and 0 in PostProcessingVolume.prefab to eliminate the issue. It's in Assets/Prefabs/

closed 2020-10-20 by EricBoiseLGSVL

asifsid-32, 2020-11-09:
> In the meantime you can set Exposure to fixed and 0 in PostProcessingVolume.prefab to eliminate the issue. It's in Assets/Prefabs/

@EricBoiseLGSVL sorry but i am not able to find Exposure in PostProcessingVolume.prefab in 2020.06 version. Is it somewhere else ?