lgsvl/simulator #947: Lidar sensor gives sigsegv

clepz, 2020-08-27:
[image]

Hi,

I've created a custom map. The map includes these:
- Terrain (It has mountains and some texture layers),
- Roads (generated with easyroad)
- Trees (used terrain brush)
- Models (cars and houses. They are lod but i use only lod0)

Unity crashes if i use lidar sensor. The screenshot above shows error. 

I can't resolve this issue. Would someone give me an advice to handle this error?

LGSVL version: 2020.05

clepz, 2020-08-28:
I found something. If I delete terrain, everything is working.
I tried that deleting my terrain and add new terrain with no changes. But I got the same error again.

I need to use this terrain, any advice?

clepz, 2020-08-28:
I found an old issue about my problem and solved the problem.

closed 2020-08-28 by clepz

SeymurSm, 2020-09-02:
Hi @clepz . Could you please share the way you solved the problem? I have same issue and can not solve the problem.

clepz, 2020-09-10:
Sorry @SeymurSm  i totally forgot to answer.
You need to convert terrain to obj. You can do it with http://wiki.unity3d.com/index.php/TerrainObjExporter.