lgsvl/simulator #948: Apollo Not Detecting Vehicles Created Using Python API

UYasher, 2020-08-28:
I am trying to create scenarios with the python API which are then navigated using Apollo. As shown in the pictures below, Apollo is not detecting vehicles in the simulator when they are created with the python API. I am not sure whether this is caused by the use of the API or by something else, but Apollo is normally able to navigate properly when I connect to Apollo manually (i.e. not with the API).

[image]
[image]

rongguodong, 2020-08-28:
Please post your Python script for us to reproduce it.

UYasher, 2020-08-28:
Here is a simple snippet which gives that behavior for me:

```
import os
import lgsvl
import sys
import time

sim = lgsvl.Simulator(os.environ.get("SIMULATOR_HOST", "127.0.0.1"), 8181)
if sim.current_scene == "SanFrancisco":
    sim.reset()
else:
    sim.load("SanFrancisco")

sim.set_time_of_day(12)
egoState = lgsvl.AgentState()
egoState.transform = sim.get_spawn()[0]
ego = sim.add_agent("Lincoln2017MKZ (Apollo 5.0)", lgsvl.AgentType.EGO, egoState)
egoX = ego.state.position.x
egoY = ego.state.position.y
egoZ = ego.state.position.z

ego.connect_bridge(os.environ.get("BRIDGE_HOST", "169.254.116.252"), 9090)

POVState = lgsvl.AgentState()
POVState.transform = sim.map_point_on_lane(lgsvl.Vector(egoX - 30, egoY, egoZ))
POV = sim.add_agent("Sedan", lgsvl.AgentType.NPC, POVState)

sim.run()
```

rongguodong, 2020-08-28:
Apollo's perception module is not stable now. I tried to manually drive the ego close to the NPC, and it can then detect it.
[image]