lgsvl/simulator #949: LiDAR horiziontal FoV

MarvinKlemp, 2020-08-28:
Is there a way to define the horiziontal FoV for the LiDAR Sensor ?
As I'm working with SICK MRS-LiDARs which only have 80° FoV 

As from the documentation I've only seen a way to change the vertical FoV

rongguodong, 2020-08-28:
Since our LidarSensor always cover full 360 degree in horizontal, there is no need to define a horizontal FoV. This is how real Lidar sensors (e.g. Velodyne Lidar) work.
However, you can make your own DirectionalLidarSensor based on current LidarSensor. The simplest way is to add one more condition [here](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Sensors/LidarSensor.cs#L272) to test if the angle is within the horizontal FoV. This will waste lots of unnecessary rendering but should generate correct result. 👍

MarvinKlemp, 2020-08-28:
Thanks a lot for the fast reply and your hard work :)!

I quickly played around without promising results.
If that's not too much work could you further elaborate how I can gain the horizontal angle of the ray from the information available in `execute()` ?

EricBoiseLGSVL, 2020-08-28:
@MarvinKlemp Are you talking about a (e.g. solid state, flash) non-spinning LiDAR?

MarvinKlemp, 2020-08-28:
@EricBoiseLGSVL yes 

Another example that actually are advertising for autonomous driving are ibeo LiDARs
For example: https://www.ibeo-as.com/en/products/sensoren/ibeoNEXTgeneric
Which has depending on the model a horiziontal resolution of 11.2 °, 32° or 60°

EricBoiseLGSVL, 2020-08-31:
@MarvinKlemp We are currently looking supporting more sensor types and we will see if we can get this or something similar on the roadmap.

MarvinKlemp, 2020-09-01:
Thanks @EricBoiseLGSVL