lgsvl/simulator #950: Question: does lgsvl has things similar to CARLA scenario runner for easy scenario creation?

AIasd, 2020-08-29:
Hi!
I am just wondering does lgsvl has things similar to CARLA scenario runner for easy scenario creation? For example, I may want to create a scenario where a leading car "gets triggered" to suddenly stop as my vehicle approaches it within a certain distance. If lgsvl supports such mechanism, could you point me to some reference on creating a scenario like this? Thank you!

EricBoiseLGSVL, 2020-08-31:
@AIasd Yes, we are currently developing a Visual Scenario Editor within the Simulator that allows users to open existing scenes, add agents, set waypoints, set triggers and support playback.  This should be releasing soon.

wisdom-bob, 2020-09-01:
> @AIasd Yes, we are currently developing a Visual Scenario Editor within the Simulator that allows users to open existing scenes, add agents, set waypoints, set triggers and support playback. This should be releasing soon.

Hello，the history of this issue is lost? Emm..... I want to ask when the curve annotation application will be published with matched map export function

EricBoiseLGSVL, 2020-09-01:
@wisdom-bob this is not that issue.  We are releasing the curve annotation exporter functionality soon

closed 2020-10-20 by EricBoiseLGSVL