lgsvl/simulator #951: Steering wheel force feedback

eyadshak, 2020-08-30:
I'm using a Fanatec steering wheel with the simulator, steering and pedal use is working fine after editing the vehicle wheel as per #155, but is it possible to add force feedback too? 
In addition, would it be possible to run Apollo with a steering wheel input in parallel?

I'm currently testing on Windows, but would also like to extend this to Ubuntu.

EricBoiseLGSVL, 2020-08-31:
@eyadshak If you add a wheel control sensor and VehicleControl sensor, you can use both at the same time.  Yes, you can implement force feedback.  We had it before in the older version of simulator, with a different wheel but it requires directly communicating with the drivers and was very specific to the hardware (plus buggy on Linux).  You will probably need to support two different implementations for the two OS'.

eyadshak, 2020-09-01:
Thanks @EricBoiseLGSVL for the answer, would you be able to guide me on how to start implementing the force feedback? In this case I will stick with Windows for that implementation before moving on to Linux if needed.

EricBoiseLGSVL, 2020-09-01:
@eyadshak Sorry, it was done before I joined the project and the engineer has since left.  I know that he created a .dll to access the driver to implement.  I found this on the Unity Asset Store that specifically mentions Fanatec.
https://assetstore.unity.com/packages/tools/integration/ginputcontrollers-multi-joystick-and-force-feedback-manager-94949

martins-mozeiko, 2020-09-02:
New Unity Input system supports haptic feedback for xbox controllers: https://docs.unity3d.com/Packages/com.unity.inputsystem@1.0/manual/Gamepad.html#rumble

So no extra packages from Asset Store are needed. Just extra code in C# to provide necessary rumble values depending on situation

eyadshak, 2020-09-06:
@EricBoiseLGSVL no problem at all, unfortunately, unfortunately, the plugin's developer has abandoned it and according to the comments force feedback isn't working properly.
@martins-mozeiko I'll have a look and try implementing it, thanks for the link.
I will post any further updates on here.

eyadshak, 2020-09-23:
The Logitech Gaming SDK on the unity store (https://assetstore.unity.com/packages/tools/integration/logitech-gaming-sdk-6630) supports implementing force feedback (tested on Windows), it works out of the box with Logitech g920 wheel and the Fanatec ClubSport Wheel Base with a minor edit.

I have implemented wheel centering and wheel control inside LGSVL, haven't tested collisions or other interactions with the environment yet, though the SDK supports them. 

Logitech G hub needs to installed, and the LogitechSteeringWheelEnginesWrapper.dll that comes with the asset is broken, so you need to download the Logitech steering wheel SDK and use the .dll that comes with it. For the Fanatec ClubSport to work, you need to edit the bindings inside SimulatorControls under scripts/controllers to correspond with the wheel.

closed 2020-09-23 by eyadshak

niezifei, 2022-08-08:
> 

@eyadshak 
Hello, may I ask how to implement Fanatec steering wheel to the Simulator? How it works?
Thank you.

EricBoiseLGSVL, 2022-08-08:
Be sure the wheel works on the PC you are developing with.  Open Unity and input debug to make sure drivers are recognized by Unity.
Clone the WheelControlSensor locally so you can see how it is made.
Open the Input Manager in Unity to assign the inputs from the Fanatec wheel.  Save the C# file it creates.
Create a bundle with the new class for the wheel control (you may need edits) and the Input file
Add to your library and assign the vehicle sensors