lgsvl/simulator #953: How to create junction when annotating the map

wisdom-bob, 2020-09-01:
I was try to annotation a map to create a map in Opendrive format. I don't know how to create a 'road' with junction features. As below is the map exported from LGSVL:
<road name="" length="22.308560132980347" id="1" junction="-1">
        <link>
            <predecessor elementType="road" elementId="0" contactPoint="end" />
            <successor elementType="road" elementId="2" contactPoint="start" />
        </link>
        ...
</road>
the road connect road="0" and road="2". the road is not a junction feature with junction="-1". However, when I export the CubeTown. I can see the junction features even when I catch the road out from the junction. As below is a road of the junction in the CubeTown:
<road name="" length="48.930436611175537" id="1" junction="3">
    <link>
        <predecessor elementType="road" elementId="0" contactPoint="end" />
        <successor elementType="road" elementId="2" contactPoint="start" />
    </link>
    ...
</road>
Connecting road="0" and road="2", the road is belongs to junction="3". 
Could you please tell me that how to set a road is belong to a junction or not?

luqiang21, 2020-09-02:
@wisdom-bob You need to place corresponding intersection lanes under an `MapIntersection` object and make sure the lane is connected with its predecessor lanes and successor lanes (i.e. their end points are overlapping).

wisdom-bob, 2020-09-02:
> @wisdom-bob You need to place corresponding intersection lanes under an `MapIntersection` object and make sure the lane is connected with its predecessor lanes and successor lanes (i.e. their end points are overlapping).

thanks, it's covered. It's my mistake...

closed 2020-09-02 by wisdom-bob