lgsvl/simulator #954: CyberRT bridge error

AlexandrZabolotny, 2020-09-01:
Hi team!

I am trying to change my sensor JSON configuration according to [this](https://www.lgsvlsimulator.com/docs/modular-testing/)  

I added the following lines 

```
  {
    "type": "3D Ground Truth",
    "name": "3D Ground Truth",
    "params": {
      "Frequency": 10,
      "MaxDistance": 100,
      "Topic": "/apollo/perception/obstacles"
    },
    "transform": {
      "x": 0,
      "y": 0,
      "z": 0,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  },
  {
    "type": "Signal",
    "name": "Signal Sensor",
    "params": {
      "Frequency": 10,
      "MaxDistance": 100,
      "Topic": "/apollo/perception/traffic_light"
    },
    "transform": {
      "x": 0,
      "y": 0,
      "z": 0,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  }
```
**I am getting the following error  
"Invalid: Unsupported message type Simulator.Bridge.Data.SignalDataArray used for CyberRT bridge"**  

[image]

daviduhm, 2020-09-01:
@AlexandrZabolotny What version of simulator are you running with? Modular testing is a new feature that we recently released on [2020.06](https://github.com/lgsvl/simulator/releases/tag/2020.06). Please use the 2020.06 simulator if you're trying to run modular testing with Apollo.

AlexandrZabolotny, 2020-09-02:
@daviduhm , Thank you for your quick response  

Thank you for your quick response

I have installed the required version.  
But now I have the following problem. As it seems to me, the navigation data is not formed correctly in the new version. This can be seen in the figures.  
I can't figure out what the problem is. Perhaps you have idea?  

**real Vehicle position**
[image]    

[image]

AlexandrZabolotny, 2020-09-02:
> It looks like you might be using Apollo 5.0 (probably from the [LGSVL fork](https://github.com/lgsvl/apollo-5.0)) in which case I think you need to update your sources to pick up the [latest apollo map changes](https://github.com/lgsvl/apollo-5.0/commit/5b9d702154df2ccaf2c52c67dc2f4b3fa81241d7) to match the updated environment bundles that are used by 2020.06.

Hi, @lemketron 
I update sources.
But the error persisted.

rongguodong, 2020-09-02:
Are you using our fork of Apollo 5.0 or Apollo master/pre6? Since the vehicle name in Dreamview is called "Lincoln2017MKZ LGSVL", I think you are using the later. If so, please choose "Mkz Lgsvl" mode.

AlexandrZabolotny, 2020-09-03:
> Are you using our fork of Apollo 5.0 or Apollo master/pre6? Since the vehicle name in Dreamview is called "Lincoln2017MKZ LGSVL", I think you are using the later. If so, please choose "Mkz Lgsvl" mode.

I've tried using both Apollo master/pre6 (select the "Mkz Lgsvl" mode) and your Apollo5.0 branch. In both cases, the error is the same.

Perhaps I did not install the new version of the simulator correctly. (2020.06). Now I'll try to reinstall it.

lemketron, 2020-09-03:
Sounds like you might need to re-download the Borregas map to get the latest one.

If you remove the data.db file from "~/.config/unity3d/LG Silicon Valley Lab/LGSVL Simulator/data.db" then when you run the simulator again it will re-download all of the maps and vehicles. See if this helps. 

Another option (especially if you don't want to lose all of your current Maps, Vehicles, and Simulations) would be to just delete or rename the current BorregasAve map on the Maps tab of the WebUI, then click "Add new" to create a new BorregasAve map. You can find the link to the [latest BorregasAve asset bundle](https://assets.lgsvlsimulator.com/51ac9efcad040833bf2eca7bdcdd90ed42aeb6bc/environment_BorregasAve) on the [LGSVL content site](https://content.lgsvlsimulator.com/maps/borregasave/).

AlexandrZabolotny, 2020-09-03:
> Another option (especially if you don't want to lose all of your current Maps, Vehicles, and Simulations) would be to just delete or rename the current BorregasAve map on the Maps tab of the WebUI, [...]

@lemketron,  thank You  
1. I followed your recommendations:  
- downloaded a latest map from the source given by you;  
- renamed it to "environment_BorregasAve_new"  
- copied to /simulator/Sim/AssetBundles/Environments   
- added a new map to http://localhost: 8080/# /Maps   

Everything is working now  

[image]