lgsvl/simulator #955: after follow the instruction to setup environment, environment cube town and vehicle file is still 0kb

iamshane, 2020-09-02:
Hello, after following the build instruction, every step is OK, and after build, the file size of  envrioment_CubeTown and vehicle_Jaguar2015XE  under the AssetBundles is 0kb,  I checked the process of build instruction, no error in the entire progress, so please help me find out, how can I make it OK, thank you.

iamshane, 2020-09-03:
@EricBoiseLGSVL 
Hello Eric, 
Thanks for your reply. I think I have found the issue, firstly the Linux support is not installed, and then the simulator build must tick both vehicle and cubetown together after I tried several times. if do this in a seperate way, the vehicle assetbundles is just the size of 30k, which is not correct.   thank you for your answer agian.

EricBoiseLGSVL, 2020-09-03:
@iamshane no problem, glad it is working.  I think you may have found a bug because any combination of vehicle/map/binary should be able to be built.  We think it's a Unity issue but we will investigate.  Thanks for the post.

closed 2020-09-03 by EricBoiseLGSVL