lgsvl/simulator #956: Simulate Data from multiple vehicle and perform collaborative positioning

weisongwen, 2020-09-02:
Hi @EricBoiseLGSVL , Thanks for sharing such a great package. I am curiously about whether the simulator can simulate camera and IMU data from multiple vehicles simultaneously?  I would like to simulate the data for collaborative positioning.

Best,
Weisong,

lemketron, 2020-09-02:
Yes, multiple vehicles are supported. If you check [How to Add/Edit a Simulation](https://www.lgsvlsimulator.com/docs/simulations-tab/#how-to-addedit-a-simulation) you will see that under 2.2: "Maps & Vehicles" the "+" button is described which will let you add another vehicle to your simulation.

Each vehicle can have its own sensor bridge (and type), specified by IP address and port.

You can also position and control multiple NPCs or Ego vehicles (and configure and connect their sensor bridges) using the [Python API](https://www.lgsvlsimulator.com/docs/python-api/).

weisongwen, 2020-09-03:
@lemketron Thanks for your fast reply. I will look into it and let you know if I have any update :-)

Best,
Weisong.

closed 2020-10-20 by EricBoiseLGSVL