lgsvl/simulator #957: BaseLink component inside the unity for creating a new vehicle?

momala, 2020-09-03:
Hi,

I followed the tutorial for creating a new vehicle for LGSVL inside the unity. I found out that there is no BaseLink component that I can add to the model like what you did in the video. I used the new release 2020.06 but there was no difference.

Where should I find this script?

Thanks

EricBoiseLGSVL, 2020-09-03:
You need to create an empty object as a child of the vehicle and add the BaseLink.cs to the object.  Then assign that transform to the VehicleSMI.cs baselink public field.

momala, 2020-09-14:
> You need to create an empty object as a child of the vehicle and add the BaseLink.cs to the object. Then assign that transform to the VehicleSMI.cs baselink public field.

Thank you for your reply,
Should be the BaseLink.cs an empty script? because I have to create it as a new component. Then I adjust the transformation on the rear axel. actually I cant find anything in the VehicleSMI.cs to assign it. Could you please explain it more?
I should mention that In the simulator the baselink is attached to the ground at the bottom of the rear axel and it's not align with it.

EricBoiseLGSVL, 2020-09-15:
Yes, add the baselink class to the BaseLink object.  You have set the public reference correctly, it just needs the transform data for VehicleSMI.  When the sensors are being setup, they need the BaseLink class to find the component.

EricBoiseLGSVL, 2020-09-15:
If you don't have this on the VehicleSMI class, you will need to download the latest 06 release.  Note, it is a branch, not master.

momala, 2020-09-15:
> If you don't have this on the VehicleSMI class, you will need to download the latest 06 release. Note, it is a branch, not master.

You are right, I didn't use the latest version, Now I update it to 06 release and everything is fine based on the tutorial. 
Thank you very much for your support.

closed 2020-09-15 by momala