lgsvl/simulator #958: How to build a 1:10 reduced size custom map

kanakab, 2020-09-03:
Hello,

I want to build or edit a map and reduce its size to 1/10th of the actual world. I want to test a RC Car (1:10) in this map?
Please tell me if this is possible in LGSVL and how to go about it??

This is similar to http://docs.donkeycar.com/guide/simulator/. I want to acheive the same in LGSVL with my own small town.

Thanks,
Abhishek

yumianhuli2, 2020-09-03:
> Hello,
>
> I want to build or edit a map and reduce its size to 1/10th of the actual world. I want to test a RC Car (1:10) in this map?
> Please tell me if this is possible in LGSVL and how to go about [...]

The map changes to one tenth of the original, and the HD map read by the algorithm also needs to be modified accordingly

EricBoiseLGSVL, 2020-09-03:
Yes @yumianhuli2 is correct for hdmap.  As for geometry, you can scale the root transform to whatever scale you prefer.  You might have issues with some meshes if they are not setup correctly

kanakab, 2020-09-04:
@yumianhuli2  @EricBoiseLGSVL Thanks for your inputs.

What about the other static actors such as traffic lights, parked vehicles, would they also get reduced to one tenth size?
Once I scale down the map, how to scale down the dynamic actors or NPCs to one tenth say pedestrians or cyclists??
Answers to these question will really kick start my project.

Thanks,
Abhishek

EricBoiseLGSVL, 2020-09-04:
@kanakab Yes you will need to scale any actors dynamically added to scene.  For the scene if the objects have different root objects, just create one and child all scene objects to it besides the annotation (That will not scale).  To scale actors, again just scale the object root or scale fbx import settings and then move all unity objects (wheel colliders etc) Be careful, Unity physics can act oddly at very low scales.  It's best you make models at a 1:1 scale like we have now and make the models to scale for the actual actor.  Why would you scale down human cyclists?  This doesn't seem like a good idea for realistic simulation.

kanakab, 2020-09-09:
Thanks for the valuable inputs. Where I can find the documetation to do this? I tried to follow as per LGSVL map annotation ( https://www.lgsvlsimulator.com/docs/map-annotation/#annotate-lanes) guideline, and scaled everything to 0.1 but it didn't work.

I also tried to find the DuckieTown Map which was earlier available in LGSVL. Currently it's not available. Is there anyway I can load the Duckietown map and  edit in LGSVL??

This is really crucial to begin my project, please guide me.

EricBoiseLGSVL, 2020-09-09:
You can find the obsolete repo [here](https://github.com/lgsvl/simulator-2019.05-obsolete).  I do not recommend using it, we no longer support.  You cannot scale annotation.  You will need to manually annotate the map.  The scale is only for the environment meshes.  Again, it is best to model to scale, not drastically reduce a large model to a smaller scale.  This will effect Unity physics.
As for examples of annotation, you can clone the CubeTown repo and see all examples we support with annotation and how to setup.  This is a great place to start.  You may need to edit some gizmo code to support such small scales so annotation points aren't too large to see what you are working on.  Hope this helps @kanakab