lgsvl/simulator #959: Aborted (core dumped)

hanyh1191, 2020-09-03:
hello,lgsvl
I have a problem that has bothered me for a few days，
I installed ubuntu and windows system on my computer, 
First, I downloaded and decompressed lgsvl.zip on ubuntu18.04, and running ./simulator will open the black window, and then flash back after a few seconds. 
Then I  followed the tutorial to run the ./simulator in the docker container successfully, but the simulator window still crashed suddenly during "Loading" (Aborted (core dumped)), is it because my Graphics card memory is only 2G?
But,  I downloaded and decompressed lgsvl.zip on windows10, I successfully run the simulator, Is this using CPU memory?
Does anyone know why this is, maybe can I make the lgsvl simulator use CPU memory on ubuntu18.04?

yumianhuli2, 2020-09-03:
> hello,lgsvl
> I have a problem that has bothered me for a few days，
> I installed ubuntu and windows system on my computer,
> First, I downloaded and decompressed lgsvl.zip on ubuntu18.04, and running [...]

GPU  required 10GB memory on one PC for apollo5.X.

EricBoiseLGSVL, 2020-09-03:
@hanyh1191 Unity HDRP requires GPU and for simulation and AD control applications on one machine requires 8-10gb mem like @yumianhuli2 posted.

hanyh1191, 2020-09-04:
> @hanyh1191 Unity HDRP requires GPU and for simulation and AD control applications on one machine requires 8-10gb mem like @yumianhuli2 posted.

Thanks, but, my computer is a dual system of ubuntu and win, I can successfully use the simulator on win, why ubuntu not? is it because I need at least 8G of Graphics card memory when using simulator on ubuntu(Aborted (core dumped))? but win does not need it.

devynliu, 2020-09-04:
hello,i met same issue.maybe you forget install "In a terminal also install libvulkan with sudo apt install libvulkan1". i have same quetion without libvulkan.hope this for your question.

EricBoiseLGSVL, 2020-09-10:
@hanyh1191 Did @devynliu help solve your issue?

lemketron, 2020-09-10:
Even if you have installed libvulkan, if you also have mesa-vulkan-drivers installed then that can cause problems.

Please take a look at (or post here) the Player.log file from ~/.config/unity3d/LG Silicon Valley Lab/LGSVL Simulator/Player.log as there are often clues in that log file.

In particular, if you see "mesa" or "Intel" mentioned in Vulkan vendor or renderer lines rather than NVIDIA, then you might have "mesa-vulkan-drivers" installed. If so, then you can remove these with `sudo apt remove mesa-vulkan-drivers`.

closed 2020-10-20 by EricBoiseLGSVL