lgsvl/simulator #960: How to simulate the Lidar laser in Unity

Jiaqing-Cao, 2020-09-04:
Hi I have run the LGSVL simulator(2020.05 version), great work.
When I am looking into the Radar and Lidar sensor part. I have a few questions:

1.The Radar senor in LGSVL simulator is using the **Collider detect** in Unity, it did not implement the Radar waves send and receive, is that right? If so, any ideas on how to simulate Radar waves send and receive in Unity ?

2.As for the Lidar part. is it **also** the Collider detect in Unity? Or it can simulate the Lidar laser in Unity? I look into the "LidarSensor.cs" and "LidarSensorBase.cs" code and did not find the simulate Lidar laser part, did I miss something?

Thank you and looking forward for you reply.

rongguodong, 2020-09-04:
LidarSensor uses depth buffer. When rendering objects into depth buffer, each ray from the camera serves as the laser ray. So it does not use Unity's collider.

Jiaqing-Cao, 2020-09-04:
@rongguodong , thanks for the reply, any idea where that part code is ?

rongguodong, 2020-09-04:
All codes are in [LidarSensorBase.cs](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Sensors/LidarSensorBase.cs) and [LidarSensor.cs](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Sensors/LidarSensor.cs).
Particularly, the rendering happens [here](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Sensors/LidarSensorBase.cs#L446). The render results are read back asynchronously, and processed in [this function](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Sensors/LidarSensor.cs#L244).

As you can see, we sampled the rendered texture. Each sample point corresponds to one laser ray.

Jiaqing-Cao, 2020-09-04:
Okay, I will look into that part, thanks a lot for your reply. 
Any idea about my first Radar question?

rongguodong, 2020-09-04:
We did not simulate the radar waves, but just use a predefined volume as a "detection region". Any NPC colliding with this volume will be detected by Radar sensor.

Jiaqing-Cao, 2020-09-04:
@rongguodong , yes, then do you have any ideas about how to simulate Radar waves send and receive in Unity?

rongguodong, 2020-09-04:
Not at this time. If this is needed by more use cases, we may include this feature in our future plan.

Jiaqing-Cao, 2020-09-04:
Okay, thanks a lot for your reply, I will look into the Radar part, If I could implement that, I am willing to contribute the code here.

closed 2021-07-16 by EricBoiseLGSVL