lgsvl/simulator #961: lgsvl 2020.06 vehicle gps data is Inconsistent with data in apollo master

devynliu, 2020-09-04:
hello,
today,i try to run 2020.06 with Apollo master.i find a question,the gps data maybe is wrong in 2020.06.
my system :Ubuntu 18.04
NVIDIA:RTX 2080Ti,drive:440.36
apollo and lgsvl run with together in ubuntu.
i make compareness with 2020.05 and 2020.06.when i use 2020.05 simulator with same sensor configure,the data in apollo master  is right.you can see the picture below.can you help slove?
the first is running with 2020.05;
the second is running with 2020.06;
[image]
[image]

devynliu, 2020-09-04:
because this "Fixed 90 degree orientation offset in GPS odometry sensor"? why make this change?

rongguodong, 2020-09-04:
Please download the latest AssetBundle for 2020.06 [here](https://content.lgsvlsimulator.com/maps/borregasave/).

EricBoiseLGSVL, 2020-09-04:
@devynliu We are very sorry for this but we had been putting off this correction for many months.  We plan on releasing a blog post very soon to explain the change and why it was necessary.

rongguodong, 2020-09-08:
@devynliu If you have the correct AssetBundle using the link above, it should work well with the HD map in Apollo master.
If you still have problem after using the latest AssetBundle, please let us know.

closed 2020-10-20 by EricBoiseLGSVL