lgsvl/simulator #963: Use LGSVL 2020.06 version with Autoware 1.13: EGO is upside down.

qmitw188, 2020-09-04:
We use the latest LGSVL 2020.06 version of the simulator with Autoware 1.13.
The EGO have an upside-down problem.
[image]

I found that the 2020.06 release notes mentioned that it supports Autoware 1.14.
Does this mean that 1.13 is not  supported anymore ?

Thanks.

EricBoiseLGSVL, 2020-09-04:
@qmitw188 Autoware 1.13 is still supported, @hadiTab can confirm.  This might be from us correcting the 90 turn issue.  Please post the unity scene with the spawn info object selected to see the orientation gizmo

qmitw188, 2020-09-08:
@EricBoiseLGSVL I uploaded the unity scene and loaded the Borregas Avenue scene. The spawn info you need is in the picture. In addition, I also uploaded the image of autoware rviz again for your analysis.

[image]

[image]

Thanks.

EricBoiseLGSVL, 2020-09-08:
Please show the spawn info object, not the ego

qmitw188, 2020-09-09:
@EricBoiseLGSVL Sorry I misunderstood your question. Here is the spawn info object:
[image]

I downloaded the Borregas Avenue asset bundle from LGSVL content server.
DL URL -  https://assets.lgsvlsimulator.com/51ac9efcad040833bf2eca7bdcdd90ed42aeb6bc/environment_BorregasAve
MD5: 44ea2ac24930f49044beba38d5b3b10c

Am I using the correct one ?

By the way, I tried the pre-built version of version 2020.06 and the git branch of release-2020.06 both have the same problem.

Thanks.

EricBoiseLGSVL, 2020-09-09:
@qmitw188 No worries, thanks.  It looks fine.  I'll need @luqiang21 or @hadiTab to help with this.  Any idea why autoware 1.13 RViz would have the vehicle on it's side?

hadiTab, 2020-09-09:
@qmitw188 We actually don't technically support Autoware 1.13.
That being said, I would expect most functionality to work. How are you running localization? Are you using GNSS for localization?
GNSS localization has some issues that have been fixed in master but are not part of 1.13.

cyberphysicalrobotics, 2020-09-09:
Hi @qmitw188 

I think autoware ai 1.14 would work. That's explained in documentation in lgsvl web site.
Could you try autoware ai 1.14?
How could you visualize vehicle model?

cyberphysicalrobotics, 2020-09-09:
@qmitw188

When I tried it on autoware ai 1.14, which is explained in: 
https://www.lgsvlsimulator.com/docs/autoware-instructions/

with the following command:
'''
$ roslaunch vehicle_description vehicle_model.launch
''''

I could see the right robot description and point clouds.
[image]

qmitw188, 2020-09-10:
@hadiTab Yes, we are using GNSS for localization.

@cyberphysicalrobotics We are trying to install and test on autoware 1.14. I'll get back to you the result when we finished the test.

Thanks.

qmitw188, 2020-09-14:
@cyberphysicalrobotics Sorry that it took so long to reply to the results. Unfortunately, we tried autoware 1.14 and the result was the same as autoware 1.13.

[image]

We used the official autoware.ai docker 1.14 vs lgsvl 2020.06 release to test, but it looks different from your test results.
Do you have any ideas what might cause this problem?

Thanks.

zerolover, 2021-02-05:
The GPS coordinate is wrong in lgsvl, see issue #1077 
You can disable gnss in ndt_matching in `my_localization.launch`
```
  <!-- ndt_matching -->
  <include file="$(find lidar_localizer)/launch/ndt_matching.launch">
    <arg name="get_height" value="$(arg get_height)" />
    <arg name="use_gnss" value="0" />
  </include>
```

EricBoiseLGSVL, 2021-02-05:
@hadiTab is this from GPS?

ZiwenWan, 2021-09-03:
Hi, I am still experiencing this problem. I am wondering whether there is a fix for this problem. Thanks a lot!