lgsvl/simulator #965: The performance gets worse with version 2020.06 when bridge is connected

gyp199009, 2020-09-08:
Hello,
I found the computer gets laggy when I changed to version 2020.06 with exactly the same map and configuration, especially when the bridge is connected while there is sensor data being transfered, the former version I used is 2020.05, what is it that influences the output performance?
Thank you in advance!

EricBoiseLGSVL, 2020-09-08:
@gyp199009 this sounds like a bug.  Please post the configuration that you are using.

gyp199009, 2020-09-09:
@EricBoiseLGSVL  
**Map:**  BorregasAve
**Vehicle:** Jaguar2015XE  with ROS bridge
**Sensors:**
[{"type": "Lidar", "name": "Lidar",
            "params": {"LaserCount": 32, "MinDistance": 0.5, "MaxDistance": 100, "RotationFrequency": 10, "MeasurementsPerRotation": 360, "FieldOfView": 41.33, "CenterAngle": 10, "Compensated": true, "PointColor": "#ff000000", "Topic": "/simulator/lidar", "Frame": "velodyne"},
            "transform": {"x": 0, "y": 2.312, "z": -0.3679201, "pitch": 0, "yaw": 0, "roll": 0}},{"type": "Color Camera", "name": "Main Camera",
            "params": {"Width": 1920, "Height": 1080, "Frequency": 15, "JpegQuality": 75, "FieldOfView": 50, "MinDistance": 0.1, "MaxDistance": 1000, "Topic": "/simulator/camera/color/compressed"},
            "transform": {"x": 0, "y": 1.7, "z": -0.2, "pitch": 0, "yaw": 0, "roll": 0}},{"type": "Depth Camera", "name": "Depth Camera",
            "params": { "Width": 1920, "Height": 1080, "Frequency": 15, "JpegQuality": 75, "FieldOfView": 50, "MinDistance": 0.1, "MaxDistance": 1000, "Topic": "/simulator/depth_camera/compressed"},
            "transform": { "x": 0, "y": 1.7, "z": -0.2, "pitch": 0, "yaw": 0, "roll": 0}},{"type": "Segmentation Camera", "name": "Segmentation Camera",
            "params": { "Width": 1920, "Height": 1080, "Frequency": 15, "FieldOfView": 50, "MinDistance": 0.1, "MaxDistance": 1000, "Topic": "/simulator/segmentation_camera/compressed" },
            "transform": { "x": 0, "y": 1.7, "z": -0.2, "pitch": 0, "yaw": 0, "roll": 0 }},{"type": "3D Ground Truth", "name": "3D Ground Truth",
            "params": { "Frequency": 10, "Topic": "/simulator/ground_truth/3d_detections" },
            "transform": { "x": 0, "y": 1.975314, "z": -0.3679201, "pitch": 0, "yaw": 0, "roll": 0 }},{"type": "2D Ground Truth", "name": "2D Ground Truth",
            "params": { "Frequency": 10, "Topic": "/simulator/ground_truth/2d_detections" },
            "transform": { "x": 0, "y": 1.7, "z": -0.2, "pitch": 0, "yaw": 0, "roll": 0 }},{"type": "Keyboard Control", "name": "Keyboard Car Control"},{"type": "Wheel Control", "name": "Wheel Car Control"}]
**System:** 
CPU:  Intel Core i5-2400 @3.1GHz
RAM: 8G
GPU: Nvidia GTX 1080
Operation system: Windows 10

EricBoiseLGSVL, 2020-09-09:
@gyp199009 Thanks for posting config.  @rongguodong can you please look at this issue also.  I'm not sure why there is a fps drop for the same config as 05 release.  It should be better from our testing.  Any help would be great.

rongguodong, 2020-09-09:
@gyp199009 Can you post your FPS numbers of both 2020.05 and 2020.06?
I just tested on my machine using your config without bridge connected. I do see 2020.05 get around 80 FPS and 2020.06 get around 70 FPS (both with all camera and lidar sensors visualized). I will test more to see what is the root cause of the difference.

gyp199009, 2020-09-10:
@rongguodong Sure, but how to get the FPS number?

rongguodong, 2020-09-10:
There are lots of free software. I used [Fraps](https://fraps.com/).

gyp199009, 2020-09-10:
@rongguodong  its around 30 FPS with 2020.05 and around 15 FPS with 2020.06 on my machine, I even can't stop running the simulation on webUI when using 2020.06, it freeses sooner after running.

rongguodong, 2020-09-10:
Did you do anything in 2020.06? Or just let it stay there doing nothing?

gyp199009, 2020-09-10:
I did nothing but run the simulation and turned on all sensors visualization, also checked the vehicles and pedestrian

rongguodong, 2020-09-10:
@gyp199009 Can you provide more details? What is the output of nvidia-smi? Do you have latest NV driver?
In your original post, you said the performance is worse when bridge is connected. Can you do more tests? What if you do not set any bridge for ego? What if you set a bridge but did not connect it? Is there any difference between ROS and Cyber bridge? Etc, etc...
These more details will help us to understand the issue better.

gyp199009, 2020-09-11:
@rongguodong 
Here is the output of nvidia-smi:
[image]
When simulator is running, system memory usage is around 26%, CPU usage is around 55%, video memory usage is less than 20%, while GPU usage is almost up to 100%.
If there is no bridge or with bridge disconnected, the FPS is around 30, and the GPU usage is around 50%.
If change the bridge to CyberRT, it seems to be much better, the FPS is around 20, and the GPU usage is bellow than 40%.
I also tested on a ego vehicle without any sensors, the FPS can be achieved up to 100.

rongguodong, 2020-09-11:
@gyp199009 Thanks for detailed data! (I think they are all for 2020.06, right? Can you also gather similar data for 2020.05?)

So for 2020.06, I summarize it as:
* Without bridge or with bridge disconnected, FPS is around 30 (which is same as 2020.05?);
* With Cyber bridge connected, FPS is around 20;
* With ROS bridge connected, FPS is around 15 or even worse (you said freeze?)
Is this summary correct? 

If so, particularly if the FPS is same for 2020.05 and 2020.06 without bridge, it seems something related to bridge, not rendering is the root cause. This would be a very useful information for us. Please do confirm this. Thanks!

gyp199009, 2020-09-11:
@rongguodong   Exactly！
And for 2020.05, it's always around 30, no matter the bridge is connected or not (same to Cyber bridge).

EricBoiseLGSVL, 2020-10-20:
@gyp199009 From our investigation, we think this is from the fixes that incorrectly had higher fps.  If you are running simulator bridge one machine with a 1080 and npcs/pedestrians, these are expected fps

closed 2020-10-20 by EricBoiseLGSVL