lgsvl/simulator #966: UML class diagram for lgsvl

dr563105, 2020-09-08:
I'd like to draw a UML class diagram for lgsvl simulator. I didn't use Python API; just simulator, sensors like colour, depth, segmentation, keyboard and radar. How would I go about making one? Especially where the data originates and where it ends up.

Appreciate any assistance. Thank you.

EricBoiseLGSVL, 2020-09-08:
@dr563105 I think the best is to start with SimulatorManager, then move down the components as they are activated.  Sensor manager, Bridge classes and then sensors, including SensorBase and CameraBase.  Just follow the bread crumbs from starting a simulation to returning the data over bridge.

dr563105, 2020-09-10:
I will look into this and reopen the issue if there are any questions.

closed 2020-09-10 by dr563105