lgsvl/simulator #968: Mapping LGSVL sensor coordinate into Blender

asifsid-32, 2020-09-09:
What is the origin for camera Sensor in LGSVL with respect to vehicle. 

If I want to calibrate the camera I have to provide coordinates into blender.

What can be done in this scenario ??

lemketron, 2020-09-09:
According to the [sensor json documentation](https://www.lgsvlsimulator.com/docs/sensor-json-options/#how-to-specify-a-sensor), the position and orientation of the sensor is defined in the sensor json parameters which are specified in the [vehicle configuration](https://www.lgsvlsimulator.com/docs/vehicles-tab/#how-to-change-the-configuration-of-a-vehicle):

> `transform` is the location and rotation of the sensor relative to the local position of the vehicle. The Unity left-hand coordinate system is used (+x right, +y up, +z forward, +pitch tilts the front down, +yaw rotates clockwise when viewed from above, +roll tilts the left side down).
> `x` is the position of the sensor along the x-axis
> `y` is the position of the sensor along the y-axis 
> `z` is the position of the sensor along the z-axis
> `pitch` is the rotation around the x-axis
> `yaw` is the rotation around the y-axis
> `roll` is the rotation around the z-axis

asifsid-32, 2020-09-09:
> According to the [sensor json documentation](https://www.lgsvlsimulator.com/docs/sensor-json-options/#how-to-specify-a-sensor), the position and orientation of the sensor is defined in the sensor json parameters which are specified in the [vehicle configuration](https://www.lgsvlsimulator.com/docs/vehicles-tab/#how-to-change-the-configuration-of-a-vehicle):
> 
> > `transform` is the location and rotation of the sensor relative to the local position of the vehicle. The Unity left-hand coordinate system is used (+x right, +y up, +z forward, +pitch tilts the front down, +yaw rotates clockwise when viewed from above, +roll tilts the left side down).
> > `x` is the position of the sensor along the x-axis
> > `y` is the position of the sensor along the y-axis 
> > `z` is the position of the sensor along the z-axis
> > `pitch` is the rotation around the x-axis
> > `yaw` is the rotation around the y-axis
> > `roll` is the rotation around the z-axis
> 
> Here is an example json configuration for a camera sensor (x, y, and z):
> ```JSON
> {
>     "type": "Color Camera",
>     "name": "Main Camera",
>     "params": {
>       "Width": 1920,
>       "Height": 1080,
>       "Frequency": 15,
>       "JpegQuality": 75,
>       "FieldOfView": 50,
>       "MinDistance": 0.1,
>       "MaxDistance": 2000,
>       "Topic": "/simulator/main_camera",
>       "Frame": "camera",
>       "Distorted": true,
>       "DistortionParameters": [
>         -0.25349, 0.11868, 0, 0
>       ]
>     },
>     "transform": {
>       "x": 0,
>       "y": 1.7,
>       "z": -0.2,
>       "pitch": 0,
>       "yaw": 0,
>       "roll": 0
>     }
> }
> ```

What can be done if we want to calibrate the camera? Is it possible in LGSVL ??

EricBoiseLGSVL, 2020-09-15:
It looks like you have multiple issues about this open #971 closing this one

closed 2020-09-15 by EricBoiseLGSVL