lgsvl/simulator #969: Connection with Socket in Simulator Build

arslansarp, 2020-09-09:
Hi ! I am trying to send some commands over TCP  connection. 
I managed to send some commands in Simulator over Unity Editor it actually worked really well. 

However, when I tried to control over simulator build version. It doesn't work.  

Here is the code snippet :

`private TcpListener tcpListener;

		/// <summary> 
		/// Background thread for TcpServer workload. 	
		/// </summary> 	
		private Thread tcpListenerThread;

		/// <summary> 	
		/// Create handle to connected tcp client. 	
		/// </summary> 	
		private TcpClient connectedTcpClient;
		private MapManager MapManager;
		public ModelsCreatorSensor creator;
		#endregion

		// Use this for initialization
		void Start()
		{
			
			// Start TcpServer background thread 		
			tcpListenerThread = new Thread(new ThreadStart(ListenForIncommingRequests));
			tcpListenerThread.IsBackground = true;
			tcpListenerThread.Start();

			MapManager = GetComponent<MapManager>();

			creator = ModelsCreatorSensor.instance;
		}

		// Update is called once per frame
		void Update()
		{
			if (Input.GetKeyDown(KeyCode.Space))
			{
				SendMessage();
			}
		}

		/// <summary> 	
		/// Runs in background TcpServerThread; Handles incomming TcpClient requests 	
		/// </summary> 	
		private void ListenForIncommingRequests()
		{
			try
			{
				// Create listener on localhost port 8052. 			
				tcpListener = new TcpListener(IPAddress.Parse("10.150.75.14"), 8052);
				tcpListener.Start();
				Debug.Log("Server is listening");
				Byte[] bytes = new Byte[1024];
				while (true)
				{
					using (connectedTcpClient = tcpListener.AcceptTcpClient())
					{
						// Get a stream object for reading 					
						using (NetworkStream stream = connectedTcpClient.GetStream())
						{
							
							int length;
							// Read incomming stream into byte arrary. 						
							while ((length = stream.Read(bytes, 0, bytes.Length)) != 0)
							{
								float tempspeed = 0;
								var incommingData = new byte[length];
								Array.Copy(bytes, 0, incommingData, 0, length);
								// Convert byte array to string message. 							
								string clientMessage = Encoding.ASCII.GetString(incommingData);
								Debug.Log("client message received as: " + clientMessage);`

EricBoiseLGSVL, 2020-09-09:
@arslansarp Can you show the usings at the top of the class? Are you using any with Editor?  What is the TCPClient?  What platform are you using?  Does this code support the platform?

arslansarp, 2020-09-10:
Somehow It actually working. I guess some bug in the build prevent it before. Thanks anyways.

closed 2020-09-10 by arslansarp