lgsvl/simulator #970: Linux build simulation error

kanakab, 2020-09-09:
Hello Team,

I am trying to launch the simulation, but I am getting the following error. All the Assetbundles are latest from the LGSVL content site. Any help appreciated.
Invalid: Out of date AssetBundle, rebuild or download latest AssetBundle.
[image]

EricBoiseLGSVL, 2020-09-09:
@kanakab Did you change the maps and vehicles to point at the new map/vehicle?  You also can try deleting the AppData folder for LG Simulator here C:\Users\YOU\AppData\LocalLow\LG Silicon Valley Lab\LGSVL Simulator.  Then run the binary we released, open the WebUI, sign in and all default maps and vehicles will download.  Then make the new simulation.  Your database might have old references.

DaPineappleDude, 2020-10-07:
> @kanakab Did you change the maps and vehicles to point at the new map/vehicle? You also can try deleting the AppData folder for LG Simulator here C:\Users\YOU\AppData\LocalLow\LG Silicon Valley [...]

Well, if you look closely OP is running it on Linux

lemketron, 2020-10-07:
> Well, if you look closely OP is running it on Linux

... in which case they should delete the `~/.config/unity3d/LG Silicon Valley Lab/LGSVL Simulator/data.db` file (or that entire directory).

closed 2020-10-07 by EricBoiseLGSVL