lgsvl/simulator #971: Camera Calibration

asifsid-32, 2020-09-09:
I want to calibrate my fish eye cameras which I have mounted on my ego cars. 

As of now I was trying to mount cameras in blender with the same mounting positions of LGSVL simulator.

But this idea also didn't work as yaw pitch and roll got changed when I used blender.

@EricBoiseLGSVL  Is there any way to easily calibrate my fisheye camera Sensors without using any other simulator !! ??

EricBoiseLGSVL, 2020-09-09:
@asifsid-32 Blender and Unity use a different coordinate system, just like ROS uses a different system.  You will need to manually convert.  [Here](https://www.youtube.com/watch?v=bpH5ZfonExo) is a video that helps explain the differences.  One way is to just model the sensors separately, export and watch the linked video, and then make the meshes apart of the vehicle prefab, not the entire mesh.  You can even make the meshes apart of the sensor prefab itself.  Hope this helps.

asifsid-32, 2020-09-10:
> @asifsid-32 Blender and Unity use a different coordinate system, just like ROS uses a different system.  You will need to manually convert.  [Here](https://www.youtube.com/watch?v=bpH5ZfonExo) is a [...]

@EricBoiseLGSVL I tried but it is too complicated !! 

When working with blender and trying to calibrate, mounting position i.e x,y,z these remain same that is not an issue !! Yaw roll and pitch get totally changed. Any simpler idea to map these values to blender ??

EricBoiseLGSVL, 2020-09-10:
[Here](https://answers.unity.com/questions/319802/rotation-when-importing-from-blender.html) is another link that has some tools for Blender.

asifsid-32, 2020-09-11:
> [Here](https://answers.unity.com/questions/319802/rotation-when-importing-from-blender.html) is another link that has some tools for Blender.

Hii @EricBoiseLGSVL can we place calib patterns in LGSVL simulator ?? I mean checker board for calibration ??

EricBoiseLGSVL, 2020-09-11:
Sure, just create a plane object, a new material, apply the checkerboard texture to the material and then apply to plane.  You can also make in blender and import into unity

asifsid-32, 2020-09-12:
> Sure, just create a plane object, a new material, apply the checkerboard texture to the material and then apply to plane.  You can also make in blender and import into unity

@EricBoiseLGSVL  

1) The camera position in lgsvl simulator is with respect to vehicle ? 

2) Position x,y,z are in meters or some other unit ??

EricBoiseLGSVL, 2020-09-15:
- make the plane centered to the vehicle at origin (0,0,0)  This is how we did earlier camera calibration for proprietary camera sensor testing and calibration.  Offset from origin if needed.  We then offset the real calibration textures to the real vehicle.
- In unity units are in meters