lgsvl/simulator #972: Error while running PythonAPI

Maddy2927, 2020-09-09:
Hey LGSVL team, I was trying to run a python API related to adding a cone which is given in the documentation. But its giving me an error. I'm pasting the error below, please help me understand and rectify this.

" ./quickstart/28-control-traffic-cone.py 
Traceback (most recent call last):
  File "./quickstart/28-control-traffic-cone.py", line 38, in <module>
    cone = sim.controllable_add("TrafficCone", state)
  File "/home/avp/anaconda3/lib/python3.7/PythonAPI/lgsvl/utils.py", line 21, in new_f
    return f(*args, **kwargs)
  File "/home/avp/anaconda3/lib/python3.7/PythonAPI/lgsvl/simulator.py", line 261, in controllable_add
    j = self.remote.command("simulator/controllable_add", args)
  File "/home/avp/anaconda3/lib/python3.7/PythonAPI/lgsvl/remote.py", line 68, in command
    raise Exception(data["error"])
Exception: Unknown 'TrafficCone' controllable prefab "

EricBoiseLGSVL, 2020-09-09:
@Maddy2927 Do you have the TrafficCone asset bundle in the AssetBundles\Controllables folder?  If not, did you clone and create the assetbundle?

Maddy2927, 2020-09-22:
@EricBoiseLGSVL Yes, I created the assetbundle in the specified folder.
Also, i wanted to know when we create our own PythonAPI files and try to run it, it gives error. Why so?
And when i run a custom API in my system and run the same file in another system, i cannot see the same output. for example, I spawned NPC and pedestrian on a particular location of a map, but this output was not shown at all when i copy paste the same file on another pc with simulator installed. Please provide some clarity on this. Thanks Eric

ntutangyun, 2020-09-22:
> And when i run a custom API in my system and run the same file in another system, i cannot see the same output. for example, I spawned NPC and pedestrian on a particular location of a map, but this output was not shown at all when i copy paste the same file on another pc with simulator installed. 

HI @Maddy2927, 

could you try to copy over the entire built simulator folder to the other pc, and try to run the python API scripts and see if they behaves the same?

On the pc where you're not able to spawn the NPC and pedestrian, are you able to run other simple API scripts like `01-connecting-to-simulator.py` and `02-loading-scene-show-spawns.py`?

EricBoiseLGSVL, 2020-09-22:
@Maddy2927 
- Do you have the latest PythonAPI pulled?
- It looks like "TrafficCone" is not found.  Do you have the asset bundle in the AssetBundles folder?
- Is the controllable named "TrafficCone" or "Traffic Cone"?  Please check the spelling
- Please see if @ntutangyun comments help.
- Are the two different PC's using the same version of simulator?  Are you using a Seed to to the simulation?  Seed to to enable determinism in the sim.
- What do you mean your created .py scripts are giving an error?  Do you have syntax errors?

Please go though this list and let us know