lgsvl/simulator #973: Localization is off and traffic light cannot be turned on when running with Apollo

AIasd, 2020-09-10:
Hi,

I am following the instruction on [Running latest Apollo with LGSVL Simulator](https://www.lgsvlsimulator.com/docs/apollo-master-instructions/) with Apollo master branch and LGSVL2020.06.

I encountered two issues:
1. localization seems off.
2.traffic light module seems to not be able to turn on.
They are shown in the following screenshot: 
[image]

I tried both the Apollo master branch today's version and a version about three months ago but the same two issues appear.

rongguodong, 2020-09-10:
This is because of the rotation of the map. Please download the latest AssetBundle of Borregas. Or you can delete the database and re-run the simulator to have it automatically download the latest AssetBundle.

AIasd, 2020-09-10:
Hi @rongguodong , thanks for your help. After I use the latest link of Borregas map and rerun the simulator, the localization issue is solved.

Also, when I switched to pre6 as you suggested the traffic light issue gets solved and the car can now follow the route successfully.

closed 2020-09-10 by AIasd