lgsvl/simulator #975: Subscribing sensor topics on ROS

User-712, 2020-09-10:
Hi Team,

I'm facing issue in listening sensors topics on ROS,

Currently am running LGSVL with following specifications:
OS:Linux 16.04 (memory: 31.3 GiB, processor: core i7, Graphics: GeForce GTX)
ROS: Kinetic
Simulator: LGSVL 2020.01 (released version)

I'm able to interface lgsvl simulator through kinetic rosbridge
1.roslaunch rosbridge_server rosbridge_websocket.launch
[image]
bridge status is connected

2. rostopic list output following topics:
[image]
able to get the sensor topics( currently am using default autoware car sensors)

3.command rostopic hz /imu_raw
[image]
but there is delay in between the topic data 

Could you please help me to figure out how to get sensor topics continuously

Is the issue is with kinetic rosbridge?

Simulator setup
[image]
[image]

json format of vehicle:
[{"type": "GPS Device", "name": "GPS",
            "params": {"Frequency": 12.5, "Topic": "/nmea_sentence", "Frame": "gps", "IgnoreMapOrigin": true},
            "transform": {"x": 0, "y": 0, "z": 0, "pitch": 0, "yaw": 0, "roll": 0}},{"type": "GPS Odometry", "name": "GPS Odometry",
            "params": {"Frequency": 12.5, "Topic": "/odom", "Frame": "gps", "IgnoreMapOrigin": true},
            "transform": {"x": 0, "y": 0, "z": 0, "pitch": 0, "yaw": 0, "roll": 0}},{"type": "IMU", "name": "IMU",
            "params": {"Topic": "/imu_raw", "Frame": "imu"},
            "transform": {"x": 0, "y": 0, "z": 0, "pitch": 0, "yaw": 0, "roll": 0}},{"type": "Lidar", "name": "Lidar",
            "params": {"LaserCount": 32, "MinDistance": 0.5, "MaxDistance": 100, "RotationFrequency": 10, "MeasurementsPerRotation": 360, "FieldOfView": 41.33, "CenterAngle": 10, "Compensated": true, "PointColor": "#ff000000", "Topic": "/points_raw", "Frame": "velodyne"},
            "transform": {"x": 0, "y": 2.312, "z": -0.3679201, "pitch": 0, "yaw": 0, "roll": 0}},{"type": "Color Camera", "name": "Main Camera",
            "params": {"Width": 1920, "Height": 1080, "Frequency": 15, "JpegQuality": 75, "FieldOfView": 50, "MinDistance": 0.1, "MaxDistance": 2000, "Topic": "/simulator/camera_node/image/compressed", "Frame": "camera"},
            "transform": {"x": 0, "y": 1.7, "z": -0.2, "pitch": 0, "yaw": 0, "roll": 0}},{"type": "Keyboard Control", "name": "Keyboard Car Control"},{"type": "Wheel Control", "name": "Wheel Car Control"},{"type": "Vehicle Control", "name": "Autoware Car Control",
            "params": {"Topic": "/vehicle_cmd"} }]

And also could you please suggest is there any other way to interface LGSVL simulator with ROS other than using  rosbridge

hadiTab, 2020-09-11:
@User-712 the performance of the Kinetic rosbridge is pretty bad and I've seen similar issues with it. Are you running it with Autoware? Since Autoware doesn't really use the camera I would suggest removing that from your sensors. That might improve things.

Overall, my first suggestion would be to try it with the Melodic rosbridge which has much better performance. If you are using Autoware, I would suggest running Autoware from docker as described in the instructions [here](https://www.lgsvlsimulator.com/docs/autoware-instructions/). The docker container uses the Melodic rosbridge.

If you must use Kinetic you could try using our fork of rosbridge available [here](https://github.com/lgsvl/rosbridge_suite). This fork is intended for use with Apollo 3.0 though and was never intended for general ROS usage so it could cause other issues.

User-712, 2020-09-14:
I removed camera and Lidar from the sensors, now am able to get imu and gps data continuously at proper rate.

thanks for your kindly help : )

EricBoiseLGSVL, 2020-09-14:
Thanks @hadiTab

closed 2020-09-14 by EricBoiseLGSVL