lgsvl/simulator #976: How to export the SingleLaneRoad map from lgsvl Apollo5.0 branch to Apollo pre6 branch?

AIasd, 2020-09-10:
Hi lgsvl team,

I am trying to run a scenario following the instruction [here](https://www.lgsvlsimulator.com/docs/api-how-to-run-scenario/). In particular, I used Apollo pre6 branch and lgsvl2020.06 version. I edited the ego vehicle to Lincoln2017MKZ (Apollo 5.0) in the VF_S_25_Slow.py script found [here](https://github.com/lgsvl/PythonAPI/blob/master/examples/NHTSA-sample-tests/Vehicle-Following/VF_S_25_Slow.py). Everything seems fine except the map `SingleLaneRoad` is not in the Apollo pre6 branch. After some search, I noticed that it is in the [lgsvl's Apollo5.0 branch](https://github.com/lgsvl/apollo-5.0/tree/simulator/modules/map/data/single_lane_road). 

My question is: how can I export the map from lgsvl's Apollo5.0 branch to Apollo pre6 branch such that I can run this vehicle following scenario properly?

lemketron, 2020-09-10:
Just copy the [map folder](https://github.com/lgsvl/apollo-5.0/tree/simulator/modules/map/data/single_lane_road) from the LGSVL Apollo 5.0 branch and drop it in "modules/map/data" in your Apollo master tree. Should work fine. 

Just make sure you are using the latest LGSVL Simulator (2020.06) along with the latest environment (map bundle). If in doubt, remove the simulator database (`$HOME/.config/unity3d/LG Silicon Valley Lab/LGSVL Simulator/data.db`) so the simulator will re-download the latest bundles when it is launched.

AIasd, 2020-09-11:
Hi @lemketron , I move the [map folder](https://github.com/lgsvl/apollo-5.0/tree/simulator/modules/map/data/single_lane_road) from LGSVL Apollo 5.0 repo to the same parent folder in Apollo pre6 and now the scenario works. Thank you for the help!

There are two things I noticed in the process:

1.I need to make modification to the position of the NPC vehicle in the scenario script `VF_S_25_Slow.py` line 44 from 
`POVState.transform = sim.map_point_on_lane(lgsvl.Vector(egoX, egoY, egoZ + 30))`
to
`POVState.transform = sim.map_point_on_lane(lgsvl.Vector(egoX - 30, egoY, egoZ))`

2.directly copy the entire map folder to overwrite the original map folder seems to result in build error as the following:
```
____Loading package: cyber/logger
____Loading package: cyber/blocker
[INFO] Building on x86_64...
[INFO] Building with --jobs=28 --ram_utilization_factor 80 for x86_64
INFO: Reading 'startup' options from /apollo/tools/bazel.rc: --batch_cpu_scheduling --host_jvm_args=-XX:-UseParallelGC --output_user_root=/apollo/.cache/bazel
INFO: Reading 'startup' options from /apollo/tools/bazel.rc: --batch_cpu_scheduling --host_jvm_args=-XX:-UseParallelGC --output_user_root=/apollo/.cache/bazel
ERROR: (09-10 23:43:32.192) /apollo/modules/map/hdmap/BUILD:5:1: no such package '@glog//': The repository could not be resolved and referenced by '//modules/map/hdmap:hdmap'.
ERROR: (09-10 23:43:32.235) Analysis of target '//modules/prediction/scenario/prioritization:obstacles_prioritizer' failed; build aborted.
INFO: (09-10 23:43:32.236) Elapsed time: 2.376s
==============================================
[ERROR] Build failed!
[INFO] Took 15 seconds
```

closed 2020-09-11 by AIasd

lemketron, 2020-09-11:
Glad it worked for you!

> 2.directly copy the entire map folder to overwrite the original map folder seems to result in build error as the following:

I updated my comment (above) to reflect that the singlelandroad map directory should be copied INTO the modules/maps/data/ directory. Hope that's more clear for anyone else who finds this answer in the future... Thanks!