lgsvl/simulator #977: Load the asset，check error

ifond, 2020-09-11:
Check after the map resource is loaded, but there are errors
reference ：https://www.lgsvlsimulator.com/docs/assets/

Checking...
ERROR: File 'Rain.vfx' with '.vfx' extension is not allowed inside '/Assets/Effects'
ERROR: File 'Snow.vfx' with '.vfx' extension is not allowed inside '/Assets/Effects'
ERROR: Folder 'HDRPDefaultResources' should not be inside of '/Assets'
ERROR: File 'ExampleFMU.fmu' with '.fmu' extension is not allowed inside '/Assets/Resources'
ERROR: File 'VFXDefaultResources.asset' should not be inside '/Assets'
WARNING: File '/Assets/Scripts/Bridge/ByteArray.cs' starts with non-ASCII characters, check if you need to remove UTF-8 BOM
WARNING: File '/Assets/Scripts/Controllers/Triggers/ITriggerAgent.cs' starts with non-ASCII characters, check if you need to remove UTF-8 BOM
WARNING: File '/Assets/Scripts/Controllers/Triggers/TriggerEffector.cs' starts with non-ASCII characters, check if you need to remove UTF-8 BOM
WARNING: File '/Assets/Scripts/Controllers/Triggers/WaypointTrigger.cs' starts with non-ASCII characters, check if you need to remove UTF-8 BOM
ERROR: File '/Assets/Scripts/Dynamics/FMU/FMU.cs' does not have correct copyright header
ERROR: File '/Assets/Scripts/Editor/FMUEditor/FMUImporter.cs' does not have correct copyright header
ERROR: File '/Assets/Scripts/Utilities/SensorPassRenderer.cs' does not have correct copyright header
ERROR: Asset '/Assets/External/Environments/SanFrancisco/SanFrancisco.unity' depends on '/Assets/Materials/EnvironmentMaterials/EnvironmentDamageAlbedo.png'
ERROR: Asset '/Assets/External/Environments/SanFrancisco/SanFrancisco.unity' depends on '/Assets/Materials/EnvironmentMaterials/EnvironmentDamageHeight.png'
ERROR: Asset '/Assets/External/Environments/SanFrancisco/SanFrancisco.unity' depends on '/Assets/Materials/EnvironmentMaterials/EnvironmentDamageNormal.png'
ERROR: Asset '/Assets/External/Environments/SanFrancisco/SanFrancisco.unity' depends on '/Assets/Models/TrafficControl/Materials/SignalTrafficVerticalAlbedo.mat'
ERROR: Asset '/Assets/External/Environments/SanFrancisco/SanFrancisco.unity' depends on '/Assets/Models/TrafficControl/Materials/SignalTrafficVerticalAlbedo.png'
ERROR: Asset '/Assets/External/Environments/SanFrancisco/SanFrancisco.unity' depends on '/Assets/Models/TrafficControl/Materials/SignalTrafficVerticalEmissive.png'
ERROR: Asset '/Assets/External/Environments/SanFrancisco/SanFrancisco.unity' depends on '/Assets/Models/TrafficControl/Materials/SignTrafficStopAlbedo.mat'
ERROR: Asset '/Assets/External/Environments/SanFrancisco/SanFrancisco.unity' depends on '/Assets/Models/TrafficControl/Materials/SignTrafficStopAlbedo.png'
ERROR: Asset '/Assets/External/Environments/SanFrancisco/SanFrancisco.unity' depends on '/Assets/Models/TrafficControl/SignalTrafficVertical.fbx'
ERROR: Asset '/Assets/External/Environments/SanFrancisco/SanFrancisco.unity' depends on '/Assets/Models/TrafficControl/SignTrafficStop.fbx'
ERROR: File name 'unity default resources' contains spaces in '/AssetBundles/steven/simulator_Data/Resources'
ERROR: File name 'No Name.mat' contains spaces in '/Assets/External/Environments/SanFrancisco/Models/SanFrancisco/StreetSignHolder/Materials'
ERROR: File name 'No Name.mat.meta' contains spaces in '/Assets/External/Environments/SanFrancisco/Models/SanFrancisco/StreetSignHolder/Materials'
ERROR: File name 'Foliage Diffusion Profile.asset' contains spaces in '/Assets/HDRPDefaultResources'
ERROR: File name 'Foliage Diffusion Profile.asset.meta' contains spaces in '/Assets/HDRPDefaultResources'
ERROR: File name 'Scene PostProcess Profile.asset' contains spaces in '/Assets/HDRPDefaultResources'
ERROR: File name 'Scene PostProcess Profile.asset.meta' contains spaces in '/Assets/HDRPDefaultResources'
ERROR: File name 'Skin Diffusion Profile.asset' contains spaces in '/Assets/HDRPDefaultResources'
ERROR: File name 'Skin Diffusion Profile.asset.meta' contains spaces in '/Assets/HDRPDefaultResources'
ERROR: File name 'Sky and Fog Settings Profile.asset' contains spaces in '/Assets/HDRPDefaultResources'
ERROR: File name 'Sky and Fog Settings Profile.asset.meta' contains spaces in '/Assets/HDRPDefaultResources'

EricBoiseLGSVL, 2020-09-11:
@ifond this check is for internal development, it should not effect anything.

closed 2020-09-15 by EricBoiseLGSVL

ifond, 2020-09-15:
@EricBoiseLGSVL However, the map cannot be compiled, showing that the generated map has a size of 0

EricBoiseLGSVL, 2020-09-15:
@ifond Do you have both Linux and Windows support modules installed in Unity Hub?  We require both since bundles are cross platform.  Please post Editor console errors or Editor player log so we can help troubleshoot.